skuep/AIOC #1: Missing license

hspil, 2022-12-24:
Genius project, Im looking forward to building some of these with my radio club. However, you havent provided a license which makes reproducing your project legally sketchy. Personally, I would love to see it published under a fairly permissive license (MIT perhaps) that would allow an east Asian manufacturer to make this into a super cheap and accessible by mass production, but that is up to you.

73

skuep, 2022-12-25:
Thanks for the praise! Between all development and design, I did not yet think of that. 

The project is destined to be open-source, hence why the PCB has an OSHW logo on it. I will see what I can come up with!

Best Regards

skuep, 2022-12-25:
Commit b5e78ce

closed 2022-12-25 by skuep