skuep/AIOC #2: Few ideas

SpudGunMan, 2023-01-02:
Be cool to make the pcb larger to accommodate a voltage path to get 12v for powering radio from usbC 

You mention sound card how's the performance with Vara? 

Plans to kit?

skuep, 2023-01-02:
I originally had the same idea with powering the radio, but it seemed to much hassle, because there is no useful "standard" for the Kenwood-K1 connector HT's. Some of them want 8.4V, some of them something else. And you would have to solder a dangling pigtail connector of the board. What maybe would be nice is, to make some kind of header available for grabbing the 5V from the board so one could feed it into a second power-supply pigtail with integrated DC/DC converter.

I have not tried Vara, in fact I never heard about it :-) Sounds very interesting!

I thought about kitting it. I think what I could do is sell a kit with the following contents
- PCB assembled with SMD components (tested and programmed would be best)
- 3d-printed soldering guide
- TRS connectors

The TRS connectors would need to be soldered to the PCB by the user. What do you think about it?

SpudGunMan, 2023-01-02:
If you pass Vara tests your kit will be popular indeed. I wouldn't rob 5v but having it there for mod ideas, could put Bluetooth or a Esp32 bridge for serial 

And leaving the TR plugs off for user install is big as it allows adaptation to any radio 

A din6 for example makes this plug into any radio not the original idea but devices like this will retrofit any older radio into winlink/vara/Direwolf combined with a pi or Linux pc you have a powerhouse!

Boort, 2023-01-02:
HackerNews came through again. :D I was JUST looking for something like this yesterday!

I received a TidRadio Bluetooth for programming my HT for Xmas. Unfortunately I don't see where it can be used for APRS/direwolf etc just programming. The deal breaker was that the app that is needed to use it is closed source and brings with a number of snooping libs and requires an email address and more permissions than I want to give it for adding repeaters while on the road. 

This looks like a fun way to get the features I want for playing with my little HT. I too would love a kit if it were available. 
If not a parts list with links to purchase the components would be helpful as I'm just getting started with DIY'ing stuff like this.

Thanks for your work and I look forward to seeing how this project comes along!
Boort

skuep, 2023-01-02:
Haha, thanks for the Hint with HackerNews! Haven't seen it yet! :-)

Funny enough, the (non-bluetooth?) TidRadio thing looks very similar to my AIOC. But yeah, the thing with snooping closed source Android Apps is just horrid, I am sorry for your bad experiences.

Note that this thing is still in Beta - mostly because I did not yet receive enough "validation" by more experienced Hams (than I am) that it is really working well with all kinds of applications and use-cases. 

The easiest to get started anyway is to use JLCPCB to order the assembled PCBs (unfortunately 5pcs minimum). I will start thinking about whether I should do Kits or not, somewhere in the beginning of this year, depending on how the Beta test goes.

A parts list/BOM is available in the repo, see ``BOM-k1-aioc.csv``below the kicad main folder.

SpudGunMan, 2023-01-02:
I got a ft70 i would like to test with, if you setup a kit of sorts post back with details!

skuep, 2023-02-04:
Is there anything still open here? Otherwise I would like to close the issue. If you have something specific, just open a new issue with a new topic.

Boort, 2023-02-05:
> Is there anything still open here? Otherwise I would like to close the issue. If you have something specific, just open a new issue with a new topic.

I received the pointer to the parts list. Been watching the project. I have no objection to closing this Ideas issue.

Boort

closed 2023-02-23 by skuep