skuep/AIOC #3: Nice Idea, Horrid Design

barnmichael, 2023-01-02:
I hope that is a prototyping test article. Plugs off the PCB is a terrible idea, especially with the USB at a right angle like that. Any slight bump to the radio or cable will likely tip the radio over and destroy the circuit board.

A much better idea would be to have two 3.5mm jacks in a straight line configuration with the USB connector, a la line lump. By using two jacks, the spacing is not near as critical. By using a variety of easily assembled interface cables, this could be used with a wide range of radios making it mush more attractive and hugely increasing the potential user base.

Just a thought.

skuep, 2023-01-02:
Thanks for your input! The idea was to use a 3d-printed mould to create an overmolded casing similar to a regular Kenwood K1 connector. 
See e.g. here: https://www.thingiverse.com/thing:2956832

This would have enough mechanical stability in my opinion, since this overmolding is used even for commercially available Kenwood K1 connectors. Also the USB-C connector is robust enough for your smartphone, so I don't think it will be an issue.

Your second point I already discussed with other people. I do not want to build another digirig. period. I wanted the PCB to be "the adapter cable" itself (that you have to buy as an accessory to your digirig). Why? Because I (and fellow makers) have ZERO issue to just order the PCB at JLCPCB. It's dead simple. Whereas building adapter cables is always a hassle. Not even speaking of sourcing X different adapter cables to sell, imagine all the tooling cost. Whereas with my "PCB-solution" it would just be another PCB design (which is easily done). Firmware is the same.

That was my design process.

skuep, 2023-01-24:
Since nothing is happening here anymore, I will close it. Housing design will be handled in a new issue :-)

closed 2023-01-24 by skuep