skuep/AIOC #4: Supplier and/or alternatives for Monacor plugs

johnboiles, 2023-01-02:
I've ordered 5pc PCBAs from JLCPCB and am now looking into sourcing the Monacor plugs. Do you have any recommended suppliers for these parts? The hits I'm seeing from a Google search [are either overpriced](https://www.amazon.com/Monacor-PG-203P-3-5-Black-Connector/dp/B003WGFH1Q) or of [uncertain reputability](https://cyberstore.ee/product/monacor-pg204p-stereo-plug-pg-204p/).

I'm super excited about this project. I made a [janky version of this kinda thing](https://github.com/johnboiles/BaofengUV5R-TRRS) a long time ago, but your iteration looks wayyyy more better!

SpudGunMan, 2023-01-02:
JAMECO, digi key, mouser, and to some degree Adafruit all have plugs. You can also get molded mic [plugs](https://a.co/d/8jsobLn)

johnboiles, 2023-01-02:
Those suppliers don't have the listed Monacor parts (or any Monacor parts for that matter). So we need to come up with some part number alternates that will work to solder directly to the PCB. If @skuep hasn't tried any other parts, I'll probably just get a few different options from Digi and experiment.

gx1400, 2023-01-03:
watching this conversation

it looks like it's just standard cheap solder connectors for cables, with the tabs bent to fit down to the copper pads.
I expect these might work:
- 2.5mm: https://www.amazon.com/Ancable-Replacement-Connector-Headphones-Headset/dp/B0785JFFN5
- 3.5mm: https://www.amazon.com/Fancasee-Replacement-Connector-Headphone-Earphone/dp/B07Y8JGFS1

I'd need to research some part numbers from digikey

For future design improvements, might suggest THT pcb mount connectors, I've used these, which I think are form, fit, function identical, 3.5 mm: 
- CUI Devices SP-3533-02 (digikey 102-4747-ND)
- Tensility 54-00036 (digikey 839-1240-ND)

johnboiles, 2023-01-03:
Good finds on both counts @gx1400! Links to those Digikey parts for future readers:
* https://www.digikey.com/en/products/detail/cui-devices/SP-3533-02/9486641
* https://www.digikey.com/en/products/detail/tensility-international-corp/54-00036/5609421

skuep, 2023-01-03:
> If @skuep hasn't tried any other parts, I'll probably just get a few different options from Digi and experiment.

I have not tried any alternatives. In Germany (Europe) you can get them at musikus-hifi.de or lautsprechershop.de.

> it looks like it's just standard cheap solder connectors for cables, with the tabs bent to fit down to the copper pads. I expect these might work:
>
>     * 2.5mm: [...]

Yes, these should work in my opinion.
 
> For future design improvements, might suggest THT pcb mount connectors, I've used these, which I think are form, fit, function identical, 3.5 mm:
> 
>     * CUI Devices SP-3533-02 (digikey 102-4747-ND)
> 
>     * Tensility 54-00036 (digikey 839-1240-ND)

Cool, I was looking for something like this in the beginning, but couldn't find it (and didn't expect them to even exist). I may have to look further!

johnboiles, 2023-01-21:
I used those Amazon parts recommended by @gx1400 and they worked great!
[image]

I've successfully sent and received APRS packets with these cables. Thanks for this project @skuep it's very fun!

skuep, 2023-01-21:
Very cool! 😎 They look good.
Programming firmware worked without issues?
Did you use the 3d printed soldering guide?

gx1400, 2023-01-21:
I used the same parts to assemble five, went together very easily.

My 3d printer is apparently over extruding so i couldn't fit the connectors into the printed fixture. I just plugged the connectors into my Baofeng UV-5R and risked heat damage. Mine mine works fine.

With windows 10, i did experience intermittent TX lock up if the RF cable to my mag mpunt antenna was too near the USB cable. Separating the cables resolved that.

johnboiles, 2023-01-21:
> Programming firmware worked without issues?

Yes! I built and uploaded without issue. The only snag I hit was it wasn't obvious to me that the repo has submodules and the resulting error message about missing headers took me a moment to troubleshoot. Maybe add a line in the README about running `git submodules update --init --recursive`.

> Did you use the 3d printed soldering guide?

Yes! I printed on an Ultimaker 3 at 0.1mm layer height and it fit perfectly! Even the little posts that go into the pcb holes fit. It was fun to use.

After posting last night I went on to try CHIRP, APRSDroid, and Direwolf. CHIRP worked mostly great, though at one point I got into a state where trying to read data from the handset caused it to reset. Some jiggling of the board and the connectors resolved it 🤷‍♂️. I couldn't get APRSDroid to work. When I'd plug in the AIOC, Android would hang for a bit. Trying to record audio from the AIOC would hang and sometimes lock up the UI. Seems like the phone was struggling to decide what to do with the AIOC sound device. Not sure what to try next. Direwolf's PTT worked great. I wasn't able to receive anything with Direwolf but maybe APRS was just quiet last night.

skuep, 2023-01-23:
> My 3d printer is apparently over extruding so i couldn't fit the connectors into the printed fixture. I just plugged the connectors into my Baofeng UV-5R and risked heat damage. Mine mine works fine.

Yeah, I suppose those whole can be a bit tight, if your printer is not completely tuned.. :-) If you have access to a 2.5mm and 3.5mm drill bit, this could be solved quite easily.

> With windows 10, i did experience intermittent TX lock up if the RF cable to my mag mpunt antenna was too near the USB cable. Separating the cables resolved that.

Interesting! Can you say what exactly locked up? I assume the USB communication somehow got disturbed. Did you have the AIOC connected directly to the PC or was there e.g. a long cable or hub in between?

> Yes! I built and uploaded without issue. The only snag I hit was it wasn't obvious to me that the repo has submodules and the resulting error message about missing headers took me a moment to troubleshoot. Maybe add a line in the README about running `git submodules update --init --recursive`.

Oooh, yeah I should add this! The easiest solution would have been to just use the binary from the GitHub Releases page :-) No need to clone and compile anything.

> Yes! I printed on an Ultimaker 3 at 0.1mm layer height and it fit perfectly! Even the little posts that go into the pcb holes fit. It was fun to use.

Nice! That sounds good.

> I couldn't get APRSDroid to work. When I'd plug in the AIOC, Android would hang for a bit. Trying to record audio from the AIOC would hang and sometimes lock up the UI. Seems like the phone was struggling to decide what to do with the AIOC sound device (opened as https://github.com/skuep/AIOC/issues/12).

My experiences with Android were also somewhat mixed, as you often cannot find out, if there is actually an error somewhere and what exactly the error is. Everything is so much hidden from the user making this really hard to debug.
There is this app called "Lesser AudioSwitch" that lets you (in the realm of what Android actually allows the user to do) to select the audio device. You could try playing around with it. 

> Direwolf's PTT worked first time. Direwolf throws an error Transmit timing error: PTT is on 2214 mSec too long. but it still worked. I wasn't able to receive anything with Direwolf but maybe APRS was just quiet last night.

That is interesting. I have yet to find out what that actually means.

johnboiles, 2023-01-23:
> The easiest solution would have been to just use the binary from the GitHub Releases page

Yes for sure! But this didn't exist when I first programmed the boards on 1/13.

> I have yet to find out what that actually means.

Same. I've opened #13 to continue investigating Dire Wolf things

gx1400, 2023-01-23:
`Yeah, I suppose those whole can be a bit tight, if your printer is not completely tuned.. :-) If you have access to a 2.5mm and 3.5mm drill bit, this could be solved quite easily.`

Good call, I did not consider that! Risking the Baofeng was pretty low risk also :)

`Interesting! Can you say what exactly locked up? I assume the USB communication somehow got disturbed. Did you have the AIOC connected directly to the PC or was there e.g. a long cable or hub in between?`

Unfortunately I don't, my guess is that the USB communication was interrupted and prevented the USB from deasserting the DTR/whatever PTT signal.  I don't honestly know how that works.

skuep, 2023-02-04:
> For future design improvements, might suggest THT pcb mount connectors, I've used these, which I think are form, fit, function identical, 3.5 mm:
> 
>     * CUI Devices SP-3533-02 (digikey 102-4747-ND)
> 
>     * Tensility 54-00036 (digikey 839-1240-ND)

Back again at this issue. I spent quite some time to find any 2.5mm (and 3.5mm) jacks that can be soldered into the PCB, since those available seem to be 3.5mm only. There is only one chinese company that I found, that seem to produce something similar:
https://www.plug-china.com/product/Male-plug-/Pcb-plug/3.5mm-pcb-plug/35-mm-stereo-60-double-tray.html
https://www.plug-china.com/product/Male-plug-/Pcb-plug/2.5-mm-pcb-plug/25-headset-plug.html

It's *dongguan dajiang electronics co., ltd.*. I tried contacting them a few days ago, but have not yet managed to get a response. I am willing to buy some connectors for a few hundred bucks, if necessary but I suspect that is not enough for them to raise their eyebrows. 

Any idea anyone?

skuep, 2023-02-26:
## Regarding Suppliers for TRS Connectors
In the context of the *Temporarily Offline* Group Buy, we found basically the same jacks at Adafruits. 
Maybe someone wants to try and confirm if those work. In my eyes they seem to work.
https://www.adafruit.com/product/1800
https://www.adafruit.com/product/1798

3166963474, 2023-10-11:
"R17 R_0603_1608Metric"What it is?Is this a resistor or inductor？

skuep, 2023-10-11:
Not sure right now, what value does it have?

rxt1077, 2023-10-20:
> "R17 R_0603_1608Metric"What it is?Is this a resistor or inductor？

@3166963474 are you inquiring because you're getting this error? #39

> ## Regarding Suppliers for TRS Connectors
>
> In the context of the _Temporarily Offline_ Group Buy, we found basically the same jacks at Adafruits. Maybe someone wants to try and confirm if those [...]

I've ordered 5 of these connectors and I'll let you know if they work.

rxt1077, 2023-11-03:
I can confirm that the Adafruit connectors work great!
[image]

skuep, 2023-11-04:
Cool, looks great! I added the Adafruit numbers to the README file.
I feel free to close this issue now. If there are any troubles in the future, feel free to reopen.

closed 2023-11-04 by skuep