skuep/AIOC #6: Documentation: clarify firmware build instructions // upload binary under "release"

gonium, 2023-01-08:
Hi,

first of all: thanks for your work, this looks interesting and I ordered a set of assembled PCBs. Now I realize that I don't understand how the firmware is build. I expected a Makefile or something, but that is not the case. Can you provide additional documentation regarding the firmware build process? 

Additionally, can you upload a binary firmware image under "Releases"? This would make it easy to get started.

Thanks,
-Mathias

skuep, 2023-01-08:
Hi, Thanks for the bump.
I want to make a first Release of the firmware via GitHub soon. Stay tuned!
If you want, you can use STM32CubeIDE (no charge and cross-platform) to import the project and start compiling. I do my best to keep the master branch always compilable.

gonium, 2023-01-09:
Thanks, I will try the STM32CubeIDE. I wasn't aware of it.

skuep, 2023-01-15:
I have added a first Release of the Firmware that you can find under the Releases of this projects.
Nothing too automated yet, but should get the job done for now.

closed 2023-01-15 by skuep