skuep/AIOC #9: Direwolf on Windows: Pre filter size of 417

gx1400, 2023-01-15:
`Dire Wolf DEVELOPMENT version 1.7 A (Mar 15 2021)
Includes optional support for:  cm108-ptt

Reading config file direwolf.conf

...

Channel 0: 1200 baud, AFSK 1200 & 2200 Hz, A+, 48000 sample rate.

'

'Calculated pre filter size of 417 is too large.
Decrease the audio sample rate or increase the decimation factor or
recompile the application with MAX_FILTER_SIZE larger than 404.
Ready to accept AGW client application 0 on port 8100 ...`

Unsure if direwolf needs to be recompiled if or if there is a setting that should be changed.

gx1400, 2023-01-15:
[image]

Ignore the binding errors, I have multiple instances of direwolf running

skuep, 2023-01-15:
Interesting! Never had this message.
Can you try a non-development Version of direwolf?
Also, you could try 24000 instead of 48000 Hz sample rate. That should also work very well.

gx1400, 2023-01-23:
Can you link to non-development version of direwolf?  

I may be able to pull down the source and compile,  but I will need to search and setup

skuep, 2023-02-23:
Was your issue solved? I would like to close this issue, feel free to reopen. But maybe you have more luck in the direwolf repository.

closed 2023-02-23 by skuep