skuep/AIOC #12: APRSDroid seems to use AIOC intermittently

johnboiles, 2023-01-21:
On my Pixel 2 XL, I couldn't get APRSDroid to work and I wasn't able to record audio to confirm the AIOC audio device was working.

When I'd plug in the AIOC, Android would hang for a bit. I tried to record audio from the AIOC on http://online-voice-recorder.com and sometimes it would hang when beginning the recording and sometimes it would just use the internal mic.

When APRSDroid would broadcast, it would sometimes come out the phone speaker even when the AIOC was connected. Any suggestions on how to debug?

skuep, 2023-01-23:
I have actually no idea how to debug the Android stuff unfortunately. Maybe I should in the meantime not advertise APRSdroid on the main page... You could try "Lesser AudioSwitch" app, in the settings there is also some debugging capabilities. Maybe something interesting in there after experiencing the intermittent problem while trying to select the AIOC audio interface?

johnboiles, 2023-01-23:
> There is this app called "Lesser AudioSwitch" 

Neat! I'll try that.

mateuszdrab, 2023-09-07:
I think there is a bug in APRSdroid https://github.com/ge0rg/aprsdroid/issues/149#issuecomment-1709169087

skuep, 2023-09-07:
Interesting, this could indeed be related! Unfortunately I haven't touched aprsdroid in a long while so I cannot really contribute a solution.

mateuszdrab, 2023-09-07:
I think the app would need some updates to handle USB microphones. I had a look online and it seems it's not quite supported well.
On the other hand, Lesser Audio Switch isn't supported on my device. Can't install it 😔

skuep, 2024-03-24:
Due to inactivity I am going to close this issue. Feel free to reopen if required.

closed 2024-03-24 by skuep

mateuszdrab, 2024-03-24:
I think this issue shouldn't be closed.
Aprsdroid still won't use any USB attached audio device which makes it unusable on phones with no audio jack.

skuep, 2024-03-24:
I think there is very little that I can actually do. It would be best if you present these problems in the above mentioned aprsdroid issue.

mateuszdrab, 2024-03-24:
> I think there is very little that I can actually do. It would be best if you present these problems in the above mentioned aprsdroid issue.

That's true...
I'm sorry I didn't realize this was the other project.