skuep/AIOC #14: PTT short on Baofeng UV-5R

yuvadm, 2023-01-25:
I'm connecting a fully built AIOC to a Baofeng UV-5R and upon full connection the PTT triggers on and does not release.

There are no shorts on either of the TRS connectors.

Is this expected?

yuvadm, 2023-01-25:
I got the 2.5 and 3.5 reversed :facepalm:

skuep, 2023-01-25:
😁 Sorry to hear! I was already thinking about what you could check and measure. There is currently a GitHub issue on a Windows System that asserts the PTT directly after plugging in. But it looks like this is not related then 👍

yuvadm, 2023-01-28:
@skuep is it possible to switch between the plugs in software?

yuvadm, 2023-01-31:
@skuep sorry for the bump and reopening, any chance I can salvage the existing board with a custom firmware fix instead of de-soldering?

skuep, 2023-01-31:
Sorry, did not see your question. No unfortunately not, since the contacts of the two connectors are very special (e.g. audio, ptt, digital).

But you can desolder the TRS connectors, by bending or pulling on the tabs with tweezers,  while desoldering. You would need however new TRS connectors

yuvadm, 2023-01-31:
Got it, thanks!

closed 2023-01-31 by yuvadm