skuep/AIOC #17: Feature:  Non-serial PTT / COS control

MiM-TMassey, 2023-02-04:
A write-up inspired by off-topic discussion from Issue #11.  I wanted to describe in detail several options for PTT besides RS-232 control lines used by other devices, as well as introduce the possibility of using AIOC more advanced applications like ASL/HamVoIP nodes and other similar full-duplex radio uses.

Currently, AIOC supports PTT control by presenting a virtual RS-232 port via USB and using status "pins" to trigger PTT (v1.0.0: DTR Low/RTS High; in the future likely just DTR High).  This is a classic PTT control mechanism in the Amateur Radio world, dating back to when computers actually had physical RS-232 ports -- and very few other ports!  Serial ports were the only convenient way to trigger a relay, which was the most straightforward way to control PTT!  Well, none of that is true today:  are there other ways to control PTT that might be both easier and more effective?

### Alternatives for PTT control
Because we're not trying to trigger relays with actual RS-232 ports, we have a lot more flexibility in triggering PTT.  Here are some alternatives:

*  VOX-based triggering

Computers are not the first time where people wanted to get away from having to activate a mechanical switch or relay just to transmit.  VOX is a very common solution to this:  when the sound level is loud enough, start transmitting.  This, though, has issues:  if the background is too noisy it can activate incorrectly; if the signal is not loud enough it may not activate properly at all; and because the sound we might want to transmit is not continuous (like pauses between letters or words), the VOX needs to keep transmitting after the silence begins because it can't be sure that the transmission is yet complete.  These issues make radio-based VOX unreliable, especially for digital signals that may need to transition from send to receive very quickly.

However, with AIOC, we have some advantages.  First of all, there is no microphone, so there's no background noise.  If noise is received from the computer, *something* intentionally generated it, so we can switch to VOX immediately.  Second of all, there is no "loud" or "quiet".  If the computer is playing sound, sound data is flowing into the AIOC -- even if a human might perceive that sound as "quiet' or even "silence".  Therefore, there does not need to be additional VOX 'hang time'.

The beauty of this is that neither the sound-generating software on the computer nor the AIOC really have any need for any out-of-band PTT-signalling mechanism.  The sound itself is a reliable signal.  On higher-end computer radio interfaces, VOX has become a common technique and has proven to be very effective.  The most common example of this is the SignaLink USB:  https://www.tigertronics.com/slusbmain.htm  This is used in numerous digital applications, including AX.25 packet, which requires fast send/receive turnaround.

This really isn't the same as traditional microphone-based VOX.  Don't think of it as VOX:  think of it as Auto-PTT.  :)  And as you have previously mentioned, Auto-PTT would allow AIOC to support applications where serial ports are difficult to use or are just not supported -- like APRSdroid.  Sound data would simply magically flow into APRSdroid, and when sound data flows out of APRSdroid, it would automatically engage the PTT.

* GPIO-based triggering

Even if there is a desire for an out-of-band PTT-signalling mechanism, there are alternatives to RS-232 signalling.  A GPIO interface can be used instead.  This is already used by software that targets the Raspberry PI; in addition, some more advanced software (such as AllStarLink) has the ability to use GPIO pins provided on CMedia CM108/119A sound chips.  In fact, modifying USB sound dongles to expose these pins and interface them to radios is one of the most common radio interfaces for AllStarLink.  (Ref: https://wiki.allstarlink.org/wiki/Radio_Connections#Modified_CM108.2FCM119_USB_FOB )

GPIO and RS-232 signalling are very similar.  However, for RS-232, those signals are only a small part of the RS-232 standard.  There's lots of other factors involved, and some of them are not under the control of the application.  At the very least, the operating system is also involved, and it may not understand -- or care -- how those status pins are being used.  With GPIO, there is a lot less going on.  These pins are designed to simply turn on or turn off (or sense if they're turned on or off), and that's exactly all we want to do with them.  Therefore, there's a lot less possibility for things to go wrong.

This technique is also commonly used in many voice and digital applications; however, these are more common in repeater controller or digital node applications, where more advanced control is needed.  See the Other control signals section below for more details.

### Full flow control:  COR/COS
PTT is a common and well understood need:  the radio is not smart enough to know when to send or not.  After all, there's sound data at the microphone all the time, so theoretically there's always data that could be sent.  The radio has to be told when there is *actually* data that needs to be sent.  And that's what Push To Talk (PTT) does:  it signals the radio when it needs to actually start sending data.

But imagine a more complex configuration such as a repeater or a node.  These can be thought of as two radios connected together, the microphone of one radio connected to the speaker of the other, and vice-versa.  Those radios may be directly connected or might be connected across the Internet, but the concept is the same either way.  In this case, there isn't just a single source of data, there are now *two* radios, both capable of sending *and* receiving data.

Just like with a single radio, each radio needs to be told when when there is data that needs to be transmitted.  That signal is PTT.  Each radio also needs a way to *tell* the other radio when it has legitimate data that it is trying to pass it along.  That signal is called COS:  Carrier Operated Switch.  (The original name was COR:  Carrier Operated Relay.  It's the same thing.)  The COS is an outbound signal on one radio that is connected to the inbound PTT of the other radio, and vice-versa.  When one radio receives a radio carrier (i.e. receives legitimate data), that radio activates the COS signal, which is passed to the other radio's PTT, which causes it to then transmits that data.  (For more details and history, see "So What's The COR or COS?? " here: https://www.repeater-builder.com/tech-info/repeater-term.html)

As you can see, because both radios can move data in two directions, both radios need both PTT and COS, to deliver signals in both directions.  And yes, this is exactly the same type of flow control you see in numerous other applications, such as RS-232 RTS/CTS or DTR/DSR!

COS is really only essential when you trying to connect two radios together, such as in a repeater or node.  This is exactly what you are doing with something like an AllStarLink node:  bridging a remote repeater to a local radio.  This literally allows you to connect to a far-away repeater and use it as if it were right in your home.  And you can do this even with a simple Baofeng and a Raspberry Pi -- if you have COS...

Unfortunately, most radios do not have a COS signal.  Because they simply receive and pass along data any time they are not actually being asked to transmit, there is no need for a specific two-way flow-control signal like COS.  If you think about it, though, radios *do* kind of have a COS signal:  that is the exact purpose of squelch!  When squelch is blocking the data, this is the same as the COS signal being inactive:  the radio is telling you "there's no legitimate data here."  When the squelch opens up and the radio passes the data along, this is the same as the COS signal being active:  the radio is telling you "I have legitimate data:  please do something with it."  In fact, this is how a COS signal is usually gotten from a radio that doesn't actually have one: go inside the radio and tap into the squelch signal!

In order for AIOC to provide a COS signal to a computer, there needs to be two elements:  A way to get a COS signal from the radio, and a way to pass it along to the computer.  Passing it along to the computer is relatively straightforward.  Most software that expects COS uses some sort of GPIO to do this.  Whatever GPIO is used to handle PTT could likely handle COS as well.  (T [... cut ...]
AllStarLink, the original worldwide network of repeaters and nodes:  https://www.allstarlink.org/
HamVoIP:  A parallel software fork that uses the same technology and network:  https://hamvoip.org/
How to connect a radio to an ASL node (N.B. Modified CM108/119A Fob):  https://wiki.allstarlink.org/wiki/Radio_Connections 
Examples of modifying Baofeng radios to connect to nodes (most connected via modified fobs):
* https://allstarsetup.com/modify-the-baofeng-888s-for-allstar/
* https://hamvoip.org/hamradio/baofeng888/

skuep, 2023-02-04:
Thanks a bunch for the extensive writeup. That's a lot for me to digest for the next days :-)

skuep, 2023-02-23:
I started with implementing basic CM108-style PTT control. I request heavy testing :-)

https://github.com/skuep/AIOC/releases/tag/v1.2.0-rc.1

skuep, 2023-02-26:
Thanks again for all the information. I have taken some time and sorting through the information.
In the information below, I am talking in the context that we now have CM108 support, i.e. full GPIO input and output is supported on the **USB Interface**.

## COS taken from inside of radio
- Currently, the GPIOs are hardcoded as outputs in the AIOC, however the current hardware version V1.0 does indeed have the possibility to *read-back* the ``PTT1`` one pin (using USART1-TX, while the UART-Transmitter is high impedance)
- However ``PTT1`` will very likely be used for actual PTTing the radio. Unfortunately, PTT2 does not have that read-back capability, but I am considering to include this in a V1.1 version. It can't hurt and is a totally backwards and forward compatible change.
- **Hardware:** COS from Radio could currently only being used, by using ``PTT2`` for PTTing the radio and using PTT1 as input. This definitely requires a totally custom wiring between the AIOC and the radio. However, getting the COS signal from the radio is already some kind of wiring effort, this could be acceptable, although high maintenance and documentation effort.
- **Software:** We would just need to make the GPIOs direction configurable, add some GPIO readout in the firmware and done. Easy. This would be transparent and would not require any configuration of the AIOC.

## COS via "reverse-VOX"
- The AIOC could try to detect if there is an audio signal at the input and report this to the USB-host. This solution still has a lot of question-marks in my head. (e.g. If the host does not "open" the soundcard (and request the sound-card to actually record data and send it to the host), the AIOC does not record any data on its ADC and could not detect anything)
- I started thinking if it wouldn't be better if the actual PC software just does the CTCSS detection and stuff like this, but you are right, that it represents some kind of symmetry if we have VOX (or **Auto-PTT**) actually in **both** directions. For RX and TX. And if there is one thing I like, than it is symmetry.
- **Software:** We would need to implement the Auto-PTT feature for both RX and TX, because when I am already at it, I would implement both at the same time. However this would require the AIOC to be somewhat configurable right? I am thinking Auto-PTT tail time, or enabling Auto-PTT at-all. The alternative would be to make Auto-PTT the de-facto default (and if people wanted, they could configure the AIOC back to the old, manual way). Both, Auto-PTT and GPIO/DTR controlled PTT would not be possible, I think. And I am a bit hesitant changing the default settings of the AIOC so drastically. I guess it boils down to: a) Would it be acceptable for AllStarLink folks having to configure the AIOC once (using a tool) to set it into the correct mode? Or b) is it better to make Auto-PTT the default. Of course your argument with the AIOC being the simplest way there is, to make an AllStarNode (assuming Duplex-Auto-PTT is the way to go) is quite intriguing...

MiM-TMassey, 2023-02-26:
I will reply in detail when I can, probably in a few days; but I want to add a really quick note for the moment.

Right now, the most common way to get a COS enabled sound interface is to break out a soldering iron.  The entire system is a raw Asterisk system with no real GUI.  The idea of a simple command line utility that you use to make some basic configuration changes once is not even in the top 10 list of hurdles to this entire process. :-). (It needs to run on Linux, and of course ideally also run on Windows; but seeing as you are a Linux guy anyway, the Linux side certainly won’t be a problem.)

Also, perfect is the enemy of good enough: as a first pass, a command line utility would be just fine. GUIs can be wrapped around such tools later. :-)

It will definitely be next week before I will have a chance, but I am very, very excited to test out the GPIO  support.

Conceptually, I’m not sure why auto PTT and DTR based PTT are mutually exclusive. I haven’t looked at the source code or anything, but as a basic state machine, it seems pretty straightforward: is DTR enabled? Raise PTT. Is audio detectable? Raise PTT. Otherwise, drop PTT.  Whether auto PTT is enabled by default or not to me should be determined by how well it works, but once it works well, I don’t see a reason why they both can’t be enabled at the same time. If you don’t touch the serial port, nothing will happen. But needing to configure it at set up one time isn’t exactly a problem.

ITCMD, 2023-05-12:
Are there any status updates for VOX / Auto PTT?

I'm making and selling these at my local club and this is the most requested feature as this would be excellent for APRSDroid. The rest of the features are good nice-to-haves but that's really the only "This is why Im not buying / making this" I'm running into.

skuep, 2023-05-13:
I have not really worked on this. I am assessing how much interest there is for it 😁 This would be the first feature where we need some way of configuring/enabling it. For this I need to implement an interface and python script to do so. 
Maybe I will have a look in the next days how much work this is

ITCMD, 2023-05-15:
I'd really appreciate it, and I can confirm there would be **a lot** of interest. This is what would bring it from a more compact USB sound card and cat cable setup to a Signalink alternative. It would be plug-and-play super easy to use and would be very valuable to many hams.

Everyone I've talked about this to, literally everyone, asked if it had vox ptt trigger.

skuep, 2023-05-18:
Got ya. I have thought about how I could go about it with my slim resources. What about the following plan:
1. Implement a simple Auto-PTT and Auto-COS using a threshold of the digital data and a time-out. Is the data above the threshold, enable PTT and reset the time-out. If the time-out expires (e.g. 1ms or 10ms), disable PTT
1. For now, release two firmware binaries, one with Auto-PTT enabled and also the "regular" firmware (PTT via serial/CM108 *only*).
1. ~Profit~ At some point in the future, set Auto-PTT as default, if it turns out, that it is more requested, than the current default (serial/CM108). And/or include a configuration feature, where you can configure the AIOC using a python script (and a currently unused HID endpoint). Enable/Disable various PTT sources and set threshold/timeout.

Whats your opinion on that?

ITCMD, 2023-05-18:
That would be super helpful, yes!

ITCMD, 2023-05-18:
How about "CATless-PTT" to show when you'd use it?

skuep, 2023-05-21:
Good news everyone, I started working on some basic Automatic PTT and COS functionality (to be renamed of course ;-)).
Since this issue is about Non-serial PTT/COS in particular, I want to move further discussion over to issue https://github.com/skuep/AIOC/issues/19.

closed 2023-05-21 by skuep

rxt1077, 2023-12-01:
I just wanted to point out that there is a simple way to do hardware COS without opening the radio on Baofeng UV-5Rs. There's a [DC bias in the speaker output](https://www.reddit.com/r/amateurradio/comments/rpo031/is_the_uv5rs_36v_dc_bias_on_the_audio_out_typical/) that you can use to trigger a GPIO pin when the squelch is open. [You can also see it used in this YouTube video](https://www.youtube.com/watch?v=QopvHGM-9_k).

I added a diode and resistor to my AIOC and then ran it to a GPIO pin on a Raspberry Pi. I imagine you could do something similar on the AIOC PCB and then return the COS status vi CM108 emulation (or some other way). I tried wiring it up on the schematic but got really stumped trying to fit anything onto that tiny little PCB. It's my first go in KiCad.

Anyway, this is what I was thinking:

[image]