skuep/AIOC #18: Firmware build instructions

techfoamer, 2023-02-19:
Are there instructions for building the firmware and the packages required to do so?  I'm thinking of some software mods.

skuep, 2023-02-20:
You can import the project into STM32CubeIDE and just build it. Should work that way!

Let me know if this works and maybe include some details or problems you encounter so I can add a section to the README file.

Thanks! 🙂

techfoamer, 2023-02-20:
So, something like this?

git clone https://github.com/skuep/AIOC.git
cd AIOC
git submodule update

skuep, 2023-02-20:
That should be it!

techfoamer, 2023-02-21:
This worked for me:

git clone https://github.com/skuep/AIOC.git
cd AIOC
git submodule update --init

...needed the --init for git to grab the tinyusb sources.  Now I can build the project!!

closed 2023-02-21 by techfoamer

skuep, 2023-02-21:
Thanks! I will update the README later :-)

ITCMD, 2023-06-19:
Trying to follow the instructions. I have STM32IDE finally installed, and when I go to import, I don't see the AIOC-fw as an option:

[image]

The AIOC-FW folder is there and populated with files, but not showing up in stm32ide.