skuep/AIOC #20: Audio works on Windows, but not MacOS and Linux

MrMinecraft97, 2023-02-25:
Issue: Audio does not work via USB cable on MacOS 10.15.7 and newest Raspberry Pi OS 64 bit. Device shows up, but program will either hang, crash or just not have audio when attempting to use as microphone input or speaker output. I have tried on mac, audacity, quicktime and the system preferences to monitor audio input and chrome to test output. On the pi I have tried direwolf for input.

Works as expected on same usb cable, same firmware and same radio with Windows 10.
Works as expected to read and write to radio using chirp on same Mac.

Current firmware: Pre-release version shipped on beta unit.

If I find anything else, I'll add it here.

skuep, 2023-02-25:
Would you be willing to update the firmware to one of the releases? I have a raspberry pi and could try to verify the issue with a known firmware.

MrMinecraft97, 2023-02-25:
I updated to 1.1.1, I'll add more after some testing.

MrMinecraft97, 2023-02-26:
After updating the AIOC to 1.1.1, direwolf 1.7 beta on the pi seems to decode most packets without issue on my KG-UV9D MATE. I made sure my direwolf.conf file was correct and I had the appropriate audio level using `direwolf -a 10`. I kept it around 50-60 but I'm not sure what a correct level should be. It didn't seem to work with the UV-5R using the same external antenna, but I don't think that is its problem as the audio is less distinct through the UV-5R.

skuep, 2023-02-26:
I think 50-60 should be alright for direwolf! Strange issue with the UV-5R.. Maybe I need to do some more testing with my UV-5R as well. Been using the UV-9D MATE a lot for APRS lately, too.

closed 2023-03-11 by skuep