skuep/AIOC #22: PTT remains on after transmitting using Direwolf on Linux

b4ux1t3, 2023-03-30:
I've tried this on both the Raspberry Pi as well as my daily driver Arch laptop.

Whenever I use /dev/ttyACM0 as my PTT for Direwolf operation, the AIOC holds open the PTT after the first transmission.

Subsequent transmissions still transmit just fine, but the only thing that closes PTT is actually unplugging the cable! Exiting Direwolf doesn't work, and re-running direwolf doesn't do anything differently

I have tested this with two (brand new) Baofeng UV-5Rs. Can test with an Explorer QRZ-1 if you think it will help (and will do so anyway later). 

As an aside, I'm noticing that the serial PTT isn't working _at all_ on Windows, but I doubt it's related directly to this issue, and need to poke at it a bit before I feel comfortable blaming that on the AIOC.

As for this issue, I'm reasonably certain it's directly related to AIOC, because (and as you can see in my config) direwolf has been working just fine with a PTT wired to its GPIO pins, and I get the exact same behavior on both Linux machines.

___

## Configs
Here's my direwolf config for Arch:

```
ADEVICE plughw:2,0 
ARATE 48000

CHANNEL 0

MYCALL XX0XXX-1

MODEM 1200
#MODEM 9600

PTT /dev/ttyACM0 DTR -RTS

AGWPORT 8000
KISSPORT 8001
```

And for the Pi:

```
ADEVICE plughw:3,0
ARATE 48000

CHANNEL 0

MYCALL XX0XXX-3

MODEM 1200
#MODEM 9600

#PTT GPIO 25
PTT /dev/ttyACM0 DTR -RTS
AGWPORT 8000
KISSPORT 8001
```

b4ux1t3, 2023-03-30:
Uh, kind of terrifyingly, _touching_ the AIOC seems to disable the PTT.

b4ux1t3, 2023-03-30:
Okay, more findings:

The aforementioned "touch it and PTT disables" behavior _only_ exhibits on my Arch laptop, which I'm using a (name-brand, known-good) USB C to USB C cable.

This _does not occur_ on the Raspberry Pi, which is using a (very cheap, but nonetheless known-good) USB A to USB C cable. No amount of touching or moving the radio causes the PTT to stop being enabled with the Pi.

skuep, 2023-04-01:
That's really odd. I have not had such an issue nor heard of it.

This seems to be related to some kind of EMI issue, if you ask me. Is the Laptop plugged-in via the power supply? For narrowing down the problem, can you try with AC charger/without? Is the Baofeng plugged into the charger or working on battery power?

Does it happen for both Baofengs, just to rule out any issue with the radio? It might be worthwhile to check continuity on the 3.5mm jack, although PTT is on the big lug on the bottom-side and I have never had connectivity issues with this contact (just with the other two contacts). Try measuring for conductivity from the PCB pad to the TRS contact (sleeve contact) while wiggling the connector carefully.

What firmware are you using? If you're running 1.2.0, can you try 1.1.1 and vice-versa?

b4ux1t3, 2023-04-01:
Will be looking at this again tonight or Monday, but I can confirm that this happens on both baofengs, and both on and off the charging cradle. I'm also on 1.2.0.

This particular laptop can't be unplugged for long anymore (it's getting old), but I'll give it a try for completeness's sake. 

I'll retest everything, though, and ensure the continuity is good!

Test cases (for my use, later) :

1.2.0
* on cradle, pi
* on cradle, laptop (plugged in) 
* on cradle, laptop (unplugged) 
* on cradle, desktop (windows) 
* off cradle, pi
* off cradle, laptop (plugged in)
* off cradle, laptop (unplugged)
* off cradle, desktop (windows) 

1.1.1
* on cradle, pi
* on cradle, laptop (plugged in) 
* on cradle, laptop (unplugged) 
* on cradle, desktop (windows) 
* off cradle, pi
* off cradle, laptop (plugged in)
* off cradle, laptop (unplugged)
* off cradle, desktop (windows)

na7q, 2023-04-08:
This can be and is a common issue with Baofeng's. Depending on the device used, it's either at the radio, or the PTT device. 

As a DIY builder of similar devices, I know the struggle.
Here's what you need to try.

Use a better antenna known to be resonant. Not the stock antenna.
Add a ferrite to the USB cable.
Add bypass capacitors to the PTT line. Sometimes 1 at the line triggering from the device, and the ptt line itself of the radio.

1 or more of these will usually fix the common issue.

TemporarilyOffline, 2023-05-02:
Baofeng's need you to be the counterpoise for the antenna - you can make a "tiger tail" and attach it to anything that connects to the shield of the antenna (usually a belt-clip screw).  I should make a video on this, but Hayden's video will have to do for now: https://www.youtube.com/shorts/ccJHdmSE7D4

b4ux1t3, 2023-05-10:
@TemporarilyOffline that's a fascinating problem here. Haha! I doubt there's anything we can do from the AIOC's perspective to resolve that problem. Thanks TO, I'll close the issue.

closed 2023-05-10 by b4ux1t3