skuep/AIOC #23: Drop in replacements for  Microchip Tech MIC5330-SSYML-TR &/or Amphenol ICC 12401832E402A?

ITCMD, 2023-04-20:
Horray for chip shortage. Im assuming this simply is not possible, but are there any drop-in replacements for either component?

[image]

ITCMD, 2023-04-20:
It's interesting that https://www.microchipdirect.com/product/MIC5330-SSYML-TR appears to have alternative MIC5330s in stock, but JLC doesn't. If those work, how hard would they be to install myself?

skuep, 2023-04-20:
In your JLCPCB profile, you can order parts from digi-key or mouser directly to JLCPCB so they can mount them for you. It just takes some time for the parts to get to china. The call it "global part sourcing".

ITCMD, 2023-04-21:
Oh awesome, thanks!

ITCMD, 2023-04-21:
And just to clarify, other MIC5330 chips should work, right?

skuep, 2023-04-21:
Ah, sorry. Yes, the MIC5330-SSYML-TR is good! Choose "cut tape" for quantities below 5000 🙂

ITCMD, 2023-04-21:
Ah ok, I was hoping the other MIC5330 chips might work too, Ill see if I can track down those via those other vendors. Thanks!

closed 2023-04-21 by ITCMD