skuep/AIOC #28: Cant update firmware on windows - Cannot open DFU device (LIBUSB_ERROR_NOT_SUPPORTED) Lost device after RESET?

ITCMD, 2023-06-19:
[image]

```
dfu-util.exe -d 1209:7388 -a 0 -s 0x08000000:leave -D aioc-fw-1.2.0-autopttcos.bin
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Opening DFU capable USB device...
Device ID 1209:7388
Run-Time device DFU version 0101
Claiming USB DFU (Run-Time) Interface...
Setting Alternate Interface zero...
Determining device status...
DFU state(0) = appIDLE, status(0) = No error condition is present
Device really in Run-Time Mode, send DFU detach request...
Device will detach and reattach...
Cannot open DFU device 0483:df11 found on devnum 32 (LIBUSB_ERROR_NOT_SUPPORTED)
Lost device after RESET?
```

Looks like it's not reconnecting as expected.
I installed the libusbk driver from Zadig. Should I be using libusb-win32 instead?

ITCMD, 2023-06-19:
Tried again with libusb-win32. Just hangs instead of failing

```
dfu-util.exe -d 1209:7388 -a 0 -s 0x08000000:leave -D aioc-fw-1.2.0-autopttcos.bin
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Opening DFU capable USB device...
Device ID 1209:7388
Run-Time device DFU version 0101
Claiming USB DFU (Run-Time) Interface...
Setting Alternate Interface zero...
Determining device status...
DFU state(0) = appIDLE, status(0) = No error condition is present
Device really in Run-Time Mode, send DFU detach request...
Device will detach and reattach...
```

skuep, 2023-06-20:
Interesting... Wonder what is going on there. Seem to be some random occurences on the internet, saying it might be related to Zadig? But nothing specific.

Does the old-school way work? I.e. manually setting the AIOC into the bootloader mode by shorting the outermost pins on the header during power-up. Does dfu-util then work?

ITCMD, 2023-06-21:
Old-school does the same thing.
Good idea for admin, I'll give that a try next time I'm able.

skuep, 2023-07-10:
Have you somehow solved the issue or know what the problem was? Or can we close this for now?

ITCMD, 2023-07-13:
Have not given it a try yet. Will update this issue once I have!

ITCMD, 2023-07-20:
thanks for your patience on this. I can confirm that running as admin fixed the issue

ITCMD, 2023-07-20:
I thought it fixed it but that's because I originally thought it passed an error. It's still not working as the device never reattaches. It also stops working until I unplug and plug it back in.

ITCMD, 2023-07-20:
I think the issue is that the libusb driver isnt set up properly. I'm looking to replace the winusb driver but Im not sure which AIOC "device" to use (theres only one plugged in) 
[image]

skuep, 2023-07-20:
On the Discord we have had now several people having issues with dfu-util under Windows (11 mostly I think?). Something seems to be borked with this combination of dfu-util+OS at some point. As a workaround currently the best way is to use a Linux machine.

I quote:
_DFU on windows is a ginormous PITA.  it's easier to boot a linux distro off of USB or something and do DFU that way_

https://discordapp.com/channels/755897811148734536/1077406707342389258/1131294896540618763

WROG208, 2023-10-14:
After a bunch of trial and error i got it to work but it downloaded the package twice saying there was an error then it completed. As soon it was programed the CMD closed. Never happened before.
Tried a baofeng and chirp but gave me an error of short header cant read radio?

StrongJohn, 2024-03-01:
### For anyone running into this problem these are my findings in windows 10;

- Without any libusb driver installed to "AIOC DFU Runtime (Interface 6)" you get `Cannot claim interface 6: LIBUSB_ERROR_NOT_SUPPORTED`.
- With libusb-win32 installed it hangs indefinitely on `Determining device status...` or `Device will detach and reattach...`.
- With libusbk installed to "AIOC DFU Runtime (Interface 6)" it works but you also need libusbk installed to "STM32 BOOTLOADER" or else you get `Cannot open DFU device 0483:df11 found on devnum 32 (LIBUSB_ERROR_NOT_SUPPORTED) Lost device after RESET?` because it cannot reconnect to the bootloader.

Now
### **How to make this work in Windows 10**

1. Connect you AIOC in bootloader mode.
2. Using Zadig install libusbk driver to `STM32 BOOTLOADER`
3. Disconnect AIOC and remove jumper then reconnect.
4. Using Zadig again install libusbk driver to `AIOC DFU Runtime (Interface 6)`
5. Now with libusbk drivers for both the Bootload and DFU installed you can follow the instructions for updating in the README.

skuep, 2024-03-24:
@StrongJohn: Thanks for the write-up, that will come in handy if someone else has issues on Windows using the dfu stuff. I need to find some place where I can sticky this information for others to easily see.

closed 2024-03-24 by skuep