skuep/AIOC #30: Add configuration interface (via HID feature reports)

skuep, 2023-06-24:
I am currently thinking about how to implement an interface to the AIOC, so that it can be customized. This will come in handy later for various planned functionalities (see e.g. https://github.com/skuep/AIOC/issues/19).

My first plan is to use HID feature reports. They are an (so far) unused feature of the AIOC HID endpoint (which was originally implemented for CM108 which I suspect will not be impacted by this addition).

The interface has to be lightweight, i.e. it has to operate with as little fuzz as possible. No installing of dubious libraries and such. I think HID will be good for that. I am planning on writing a simple python CLI application (maybe with some abstraction library for others to use for non-CLI stuff) to control the AIOC. I am planning to use the python package "hidapi", it should work on Windows, Linux and macOS.

For now, I would like anyone interested to run this little test script on their hardware. Please post the output of the script and add information on what OS you are using. This test requires AIOC v1.2.0 firmware.

Install hidapi for Python:
``python -m pip install hidapi`` (or ``python3`` instead of ``python`` e.g. for Linux)

Test script:
````
import hid

# Open device
# In the future we would need to check if multiple AIOCs are connected. 
# In that case we require an additional serial number to select the correct device.
aioc = hid.Device(vid=0x1209, pid=0x7388) 

# Print information. Should output
# Manufacturer: AIOC
# Product: All-In-One-Cable
# Serial No: xxxxxxxx
print(f'Manufacturer: {aioc.manufacturer}')
print(f'Product: {aioc.product}')
print(f'Serial No: {aioc.serial}')

# Try to get feature report from AIOC. This is not implemented on the AIOC (yet), so it will return an empty report with 0 length
# b'\x17'
print(aioc.get_feature_report(23, 8))
````

ITCMD, 2023-09-07:
Wanted to throw a note here requesting the configurability of the PTT2 pad in the interface as per #34. Thanks!

skuep, 2023-09-10:
Alright, have a look at this PR:
https://github.com/skuep/AIOC/pull/37

I tried implementing your feature request as a first example. Firmware binary is included and a sample Python script on how to change the AIOC settings.

skuep, 2023-12-30:
Merged to PR, will be implemented in the next release. PTT2 will be able to be configured as shown in the PR. I will close this issue for now, you can re-open this issue if you run into troubles.

closed 2023-12-30 by skuep

W1BTR, 2025-06-06:
@skuep - wanted to note, you instruct the installation of hidapi here, I believe that's the one that _doesnt_ work, right? python3-hid would be the right one in that case