skuep/AIOC #31: Mirrored Wiring Firmware Support?

ITCMD, 2023-07-04:
Since RFI is an issue with the cable going parallel to the antenna, would it be possible to support having one wired up in reverse?

[image]

skuep, 2023-07-10:
Unfortunately not. It would require a new PCB layout, since some special pins can not be just swapped (e.g. analog in/out or ground pin)

ITCMD, 2023-07-13:
Makes sense. Perhaps a reversed image would be helpful

skuep, 2023-07-16:
I had a few thoughts about it and I don't think I can currently do it easily as it requires a lot of hardware changes and I don't have a lot of spare time at the moment. A solution however could be to use a K1 connector extension cord?

ITCMD, 2023-07-16:
Good idea for a workaround, that would likely work. Perhaps something like this: https://www.ebay.com/itm/303896324294

ITCMD, 2023-07-19:
Unless you have any other thoughts, I'll close this one out for now!

closed 2023-07-19 by ITCMD