skuep/AIOC #32: What is the TTL pinout for the AIOC?

ITCMD, 2023-07-13:
Trying to make TTL cables for other radios. Right now the 778UV by Anytone
https://www.avalonarc.org.uk/2019/05-12-how-to-make-a-programming-cable-for-the-anytone-778uv.html
[image]

I assumed the pinout was like this for it to go with a UV-5R:
[image]

So I shorted the RX and TX data together, hooked that up to the RX&TXData pin for the 778UV, and attached GND, but it gets no response from the radio. 

Any tips?

ITCMD, 2023-07-13:
I was able to make this work with another usb to tty device with the same pinout, fyi.

skuep, 2023-07-16:
Hm, Interesting! I am currently not sure if this works as you want it to. I think the Push-Pull output of the TX pin needs to be High-Z when nothing is being sent to the radio via the AIOC. I need to check to see if that is actually the case.

What software and what firmware version are you using, btw? There could be another issue, where asserting DTR or RTS (one of those) will assert the PTT line, which is on the same pad as the TTL RX from the radio. This could be easily tested by using the Virtual-PTT firmware I uploaded from the other issue. Here, RTS/DTR PTT should be completely disabled and not interfere with the serial stuff.

ITCMD, 2023-07-16:
Not entirerly sure how it's used. Serial communication is beyond me for the most part and Im not sure what pushing or pulling is in that context.

The software I'm using is CHIRP, which is the goto for programming radios. It works perfectly with kenwood-jack style radios.

ITCMD, 2023-07-16:
I tried it with the virtual-ptt firmware and same exact thing happens. The red TX light flashes once (a very short flash) and chirp immediately says:

ITCMD, 2023-07-16:
Out of curiosity, what does high-z mean? Which pin specifically is the PTT pin just to make sure we're on the same page?

skuep, 2023-07-30:
> Not entirerly sure how it's used. Serial communication is beyond me for the most part and Im not sure what pushing or pulling is in that context.
> 
> The software I'm using is CHIRP, which is the goto for programming radios. It works perfectly with kenwood-jack style radios.

The problem I am thinking of is essentially, that when you short RXD to TXD and connect that to the radio, it is possible that there are two "signal drivers" on a single line. I.e. the PC would need to know when to "disable" the TXD pin (i.e. make it appear as such the radio does not see the transmitting PC anymore) in order for the radio to "use" the single line for sending data. The only explanation for why it even works with your USB-TTL adapter (at-all) is, that there are series resistors on the USB-TTL module.

Now, having said that, the AIOC behaves as such, that it enables it's TXD output driver only, while it actively sends serial data over the line. So in theory, the AIOC should already be quite suitable for such a single-line-TTL scenario. The problem must be somewhere else.

> Here's how I have it wired (I have checked continuity) with orange/white going to ground and brown going to the TX and RX pins both shorted directly together

From what I can tell, that looks okay. I am not so sure if there is a orange/white and brown/white mixup? Comparing the images to your text. Can you verify that with the same wiring, it works using the other USB-TTL adapter?

> From my understanding, these radios use half-duplex serial communication, which is what needs to be added to the AIOC somehow. In half-duplex communication, data is transmitted in one direction at a time, and the transmitting and receiving devices take turns sending and receiving data.

Yes, exactly, that's what it looks like. As I said, theoretically the AIOC should be already able to handle that scenario (because this feature is integrated due to the TX being shared with the PTT).

> I wonder if the unused 3.5mm tip could be used as a secondary input or something in the firmware. It'd also have to be ready to accept the +3.3 / +5v from some kenwood jack radios.

Currently, it can only be used as an output (PTT2) and not read back by the microcontroller. If there is demand, I have a draft revision adding this readback functionality into the firmware. But so far I don't see a clear way to include that feature in the firmware (i.e. where to report the status of this pin?) so I didn't bother publishing it.

In my opinion, the AIOC should work in this scenario. To narrow down the issue I propose
- Connect RXD and TXD of the AIOC (and nothing else). Use a terminal and see, if the loopback works, i.e. do you receive back the characters you type? This should definitely work, as I have tested this now.
- Add radio serial line to the shorted RXD/TXD pins. Does it work? If you have a hot air station and are not shy of fiddling around with the AIOC, you could try to temporarily remove R13. This will remove the PTT "output" from the serial line, in case it is still the issue.
- Don't know any further what to do.

ITCMD, 2023-07-31:
Ill do some testing and re-read some of this when it's not after midnight, but typically duplex communication is handled by the chip, not the computer. For example, when using the standard cable that has tx and rx shorted, there is no feedback from what is sent out. The chip acts as an intermediary agent and has it wait and go. Not certain on this, however, so I'll have to look more into it.

No oscilloscope to check during low phases but can check on idle!

tandersson, 2023-09-14:
Hi,

Had a similar TX/RX issue with the CRT Micron, which probably use the same HW as AT778-UV and RT95.
Found that by adding silicon diode to the TX I was able to pull TX low which worked better with interfaces that use a pull-up resistor on RX. See attached schematics.

[image]

skuep, 2024-03-24:
@ITCMD I think @tandersson has a good solution to your issue for now, right? I assume I can close the issue for now. If there is anything I missed, feel free to reopen.

closed 2024-03-24 by skuep

ITCMD, 2024-03-28:
> @ITCMD I think @tandersson has a good solution to your issue for now, right? I assume I can close the issue for now. If there is anything I missed, feel free to reopen.

Sorry for the silence lately! Moved, graduated, married, new job... lots of balls to juggle.

Still catching up on shipping AIOCs sold before implementing some changes.

That looks like a great fix. Thank you @tandersson for the help!

I'm learning more every day about all of this, interesting point on the 3.3 vs 5v.

skuep, 2024-03-29:
Wow, congrats! I am also very busy lately, even though I only have barely half of your stuff on my platter..

Just wanted to get a better overview on who needs immediate help or attention from me. If you do, just reopen the issue.

Best regards

W1BTR, 2024-07-26:
Bit thrown off by this post. @skuep - can you confirm if by default the serial output from the AIOC is at 3.3v or 5v? There are a few discussion posts warning that the 5v serial output from AIOCs could be damaging, but looking at the schematic Im wondering if it is just at 3.3v output (in which case I need to remove the voltage divider from the schematic)

skuep, 2024-07-26:
The AIOC is all 3.3V, you won't damage anything with that.. in addition the AIOC RX input is 5V capable, so no damage to the AIOC either.

W1BTR, 2024-07-26:
> The AIOC is all 3.3V, you won't damage anything with that.. in addition the AIOC RX input is 5V capable, so no damage to the AIOC either by 5V radios.

TY for the quick reply

I knew the RX supported 5V, was thrown off by the 3.3v. That's pretty hilarious, I'll have to correct that in the video as well. Luckily I discovered TXD can be restored by removing the two resistors. V 1.2 will have that fixed, haha

skuep, 2024-07-26:
Hardware is hard 😁