skuep/AIOC #34: PTT Triggered by DATA with cat control

ITCMD, 2023-07-19:
I wired up my FT-857D to an AIOC. It works well. I can use it like a digirig with audio and PTT, and I can also do cat control and programming. However, I cant do cat control and audio in and out at the same time. If I have both cables plugged in at once, it constantly toggles PTT over and over again for very short intervals (same speed as the TX of data).

[image]

Is this expected behavior? Is it possible to do something like CAT control while also using the soundcard and PTT features of the AIOC perhaps by adding another component in my wiring or by rearranging the wires some?

ITCMD, 2023-07-19:
Probing with the multimeter, PTT isnt shorted when TXing serial data, but when it's not transmitting serial data (to the radio) the voltage is 3.3 volts across PTT. When it transmits, the voltage drops to 1.5v. Perhaps a capacitor could help with that?

ITCMD, 2023-07-21:
Wanted to add, this happens on both the standard and virtual ptt firmware.

ITCMD, 2023-07-24:
Tested this on a few more radios (FT-100d, 878d, Xiegu G90) and am having the same issue. When any data communication goes on, it triggers PTT.

ITCMD, 2023-07-26:
Sorry to bug ya! Just wanted to add some activity here in case this issue got lost in the inbox. You're likely just busy with things which I _totally_ get. @skuep

skuep, 2023-07-27:
Not lost at all! Just a bit swamped with work and health currently. I am planning to look into it this or next weekend.

ITCMD, 2023-07-28:
Gotchya, thanks so much, Simon! I certainly get that.

skuep, 2023-07-30:
> Sorry for yet another issue to respond to! I know you're busy and I hate taking up your time
> 
> I wired up my FT-857D to an AIOC. It works well. I can use it like a digirig with audio and PTT, and I can also do cat control and programming. However, I cant do cat control and audio in and out at the same time. If I have both cables plugged in at once, it constantly toggles PTT over and over again for very short intervals (same speed as the TX of data).

Okay, so we are now getting into the hacky territory? :-) That's not a HT right? That's a "real" radio, that you can remotely control (CAT) and has audio in/out. I like this, and I think the AIOC should be able to work with something like this. The AIOC has been designed with HT's in mind, so that means the serial port and PTT can only work exclusively (because they are on the same pad essentially).

> [image]
> 
> Is this expected behavior? Is it possible to do something like CAT control while also using the soundcard and PTT features of the AIOC perhaps by adding another component in my wiring or by rearranging the wires some?

If I understand you correctly, that is the (current) expected behaviour. Due to the PTT line being connected to the TXD (Radio RXD) line, the firmware will clear any PTT when it receives a "write serial data" command from the PC. That alone should probably be an issue or is it? I mean, if the radio only sends serial data during transmit silence (PTT off) it should be fine, right?

However, the second (your) problem is, that the PTT and Radio-RXD are shorted on the AIOC. Thus, each time the PC sends some data, the PTT pin state **will** change!
Currently, I can see the following solution:
1. Remove R13, this will remove the PTT from the serial line. The problem should be gone, but you won't have any PTT anymore
1. Connect your PTT wire directly to one of the pads of where R13 (see image) was

For a better solution, I could offer you the PTT2 pin (see image). That would require some firmware changes to make it configurable, that the AIOC uses PTT2 pin instead of PTT(1). Since I have configurability of the AIOC planned for the near future (using a simple python tool), this could be a good solution.

ITCMD, 2023-07-31:
All good ideas. Currently leaving PTT disconnected is my solution, since CAT control can work to trigger PTT, although I'd like to have the ability to hotswap between rig control ptt and AIOC's ptt without having to flip a switch or solder or something (as some apps dont support rig control).

Removing R13 and connecting to that PTT pad in red, that would isolate PTT from the rest? That sounds like a good temporary solution.

The firmware to use PTT2 would be excellent down the road once the python script is done more.

As far as MIC ground connected there, that's what is expected for the official kenwood jack connection standard. I think it works because it gets shorted when PTT is triggered, which it always would be for most MIC input... hmmm

For reference, I based this on:
[image]
and
[image]

I think I had one customer wire it to the same ground as speaker and it also worked, so that might be better for compatibility.

Thanks so much for your help!

skuep, 2023-09-05:
I think issue https://github.com/skuep/AIOC/issues/30 is already a good start. You can consolidate your feature request proposal regarding configurability of PTT2 in that issue.

I am sorry that I have not yet gotten around to do it. Unfortunately until mid of September I am still not able to, due to work travel. Thanks for your offer anyway but I think I would rather remain independent :-)

ITCMD, 2023-09-07:
> I think issue #30 is already a good start. You can consolidate your feature request proposal regarding configurability of PTT2 in that issue.
>
> I am sorry that I have not yet gotten around to do it. [...]

All totally understandable, and no rush. Just figured if it helped I wouldnt mind throwing support your way with no strings or expectations. I pray work and travel goes well!

skuep, 2023-11-04:
Have you had another go at the feature? Would be interesting to know if changing to the PTT2 setting works using the HID Python script.

ITCMD, 2023-11-06:
I have not, missed your comment on the phython script! I'll have to take a look and figure out how to use it in the coming weeks!

skuep, 2023-12-31:
Good news, the next hardware revision will have a split-pad for the PTT1/RX signal so you can actually use both signals separately, if you're not using the K1 layout.
Also, PTT1 and PTT2 will be input-capable as well.

[image]

skuep, 2024-03-24:
I will close this issue for now. I will let everyone know when the next hardware revision arrives.

closed 2024-03-24 by skuep

ITCMD, 2024-06-01:
Hi @skuep,

Picking up on this project again. Trying to refresh my memory on some of this.

Where can I download the second hardware revision? Id like to take advantage of the split ptt and serial pads on my next big order.

I also hope to customize the design a bit to add some resistors and bring the output serial voltage down to 3.3v on an optional pad.