skuep/AIOC #36: Build with latest STMCube 1.13.1 fails

pbilke, 2023-08-17:
Basically the library shipped with this version of the IDE produces warnings that fail the build.   Things that were empty stubs before now indicate they are empty.   I made build by adding a source file that reproduces the old behavior.   Not sure what the maintainers want to fix these but I just added a nonimpl.c source file with this following:
/*
 * NotImpl.c
 *
 *  Created on: Aug 17, 2023
 *      Author: paul
 */

void _close(void)
{
}
void _lseek(void)
{

}
void _read(void)
{
}
void _fstat(void)
{
}
void _getpid(void)
{
}
void _isatty(void)
{
}
void _kill(void)
{
}

skuep, 2023-08-17:
Thanks for the heads-up! I will put it into my Todo list, I am currently very busy unfortunately so it may take a few weeks.

skuep, 2023-09-10:
Confirmed working now, thanks again!

closed 2023-09-10 by skuep