skuep/AIOC #38: Windows 7,8 drivers

stev1974, 2023-10-11:
Is there a way to make AIOC work with Windows 7 or 8?
I tried about a million drivers without luck.
Hope someone can give me a hint.

[image]

[image]

skuep, 2023-10-11:
The AIOC uses built-in drivers (USB standard drivers). I am not sure, why your machine is not loading them. Is there an error Message somewhere?

stev1974, 2023-10-11:
There stands if I translate to english "There were no driver for the device installed (Code 28)".
This is on Windows 8.1. I tried it also on my laptop with the same result and a friend of mine on Windows 7. Everywhere the same.
[image]

Then I tried to find USB class 2 Audio drivers and manually installed them to the device but nothing worked so far.
On my tablet with windows 10 it found the drivers without problems.

skuep, 2023-10-11:
It may be possible that the Version of Win 7/8 is too old, for installing these automatic USB drivers, but I am not sure. But since there are no updates any more for those Windows versions, I am not sure how you can make sure, you have the latest Win 7/8.. Hm

stev1974, 2023-10-11:
yes it seems like. If someone has an idea what driver I can try please let me know.

skuep, 2023-12-30:
I am afraid if nobody else has an idea, you are out of luck. From my side, any OS that doesn't support the USB Audio Class 2.0 codec is unsupported due to time limitations.
So I am going ahead and close this issue. If you have new information, feel free to re-open.

closed 2023-12-30 by skuep

OK2MOP, 2024-07-26:
For future reference, the working serial COM port driver for Windows 7 is https://www.st.com/en/development-tools/stsw-stm32102.html

I was, however, not able to find an audio and DFU driver which would match yet.

skuep, 2024-07-26:
How did you install the STM32 driver? I think it's for a totally different USB VID:PID pair belonging to STM32.
This is not the correct driver, although it looks like it's working. 

The correct driver's are the windows built-in CDC, UAC2 and HID drivers.