skuep/AIOC #43: Request to Update PCB Design for JLCPCB Assembly Including TRS Connectors and Header

alper061, 2023-11-08:
## Issue Title
Request to Update PCB Design for JLCPCB Assembly Including TRS Connectors and Header

## Description
Hello,

I am interested in ordering the AIOC PCBs from JLCPCB with a fully assembled option, including the TRS connectors and the header. The current design and accompanying files (BOM and CPL) do not seem to include the Monacor PG-204P and PG-203P TRS connectors or a header that is required for the complete assembly.

## Request
Would it be possible to update the PCB design files, Bill of Materials (BOM), and Centroid (CPL) files to include these components? This would allow for a complete assembly by JLCPCB without the need for manual soldering of these connectors post-delivery.

The specific parts that I am referring to are:
- Monacor PG-204P and PG-203P or compatible TRS connectors (2 solder lugs and a big tab for the sleeve connection)
- Any necessary headers for programming and other functionalities

## Additional Context
The Adafruit products 1800 and 1798 are confirmed to work as compatible TRS connectors. Including these in the PCB assembly would greatly simplify the process for end-users who wish to have a ready-to-use board upon delivery.

## Conclusion
Having a fully assembled board directly from JLCPCB would be incredibly beneficial for users who may not have the equipment or skills to solder these components themselves. I believe this update would make the AIOC even more accessible to a broader audience.

Thank you for considering this request. I am looking forward to your response and am happy to discuss this further if needed.

Best regards,
Alper061

alper061, 2023-11-08:
Hello @skuep,

I understand that the Monacor connectors are not suitable for machine placement by JLCPCB. Considering this, I would like to inquire whether there are alternative TRS connectors available that would be compatible with JLCPCB's automated assembly processes.

Additionally, if no such machine-placeable connectors are available, could JLCPCB consider offering a hand-soldering service for these specific components as an optional service, even if it comes at a higher cost? This would enable an optional "fully assembled" version to be offered alongside the more cost-effective, nearly assembled option currently available.

Providing a fully assembled board directly from JLCPCB would be a significant value addition for users like myself who are visually impaired and unable to solder, as well as for others who might prefer a ready-to-use board upon delivery.

I believe that offering such options would greatly enhance the accessibility and appeal of the AIOC to a broader audience.

Thank you for considering these possibilities. I am looking forward to your response and am open to discussing this further if needed.

Best regards,
Alper061

alper061, 2023-11-08:
> Thanks for the request. Unfortunately the Monacor connectors are actually
> not a PCB-mounted component, thus JLCPCB can not load them into their
> machines and place them onto the board.
>
> They need to be soldered by hand, which JLCPCB does not offer (at least not
> in this price tier).

I found some connectors:
https://jlcpcb.com/parts/2nd/Connectors/Audio_Connectors_79431

I found the another compatible connectors from Adafruit Industries,
but its not tagged properly, so i need to search directly:
https://jlcpcb.com/partdetail/AdafruitIndustries-1798/C4992016
https://jlcpcb.com/partdetail/AdafruitIndustries-1800/C4992105

These are supporting SMT Assembly.

alper061, 2023-11-08:
I found some connectors:
https://jlcpcb.com/parts/2nd/Connectors/Audio_Connectors_79431

I found the another compatible connectors from Adafruit Industries, but its not tagged properly, so i need to search directly:
https://jlcpcb.com/partdetail/AdafruitIndustries-1798/C4992016
https://jlcpcb.com/partdetail/AdafruitIndustries-1800/C4992105

These are supporting SMT Assembly.

skuep, 2023-11-18:
Where is it saying that they support and SMD assembly? As far as I can tell they are exactly like the monacor connectors and are designed to be soldered to a cable.
I highly doubt jlcpcb will solder these on by hand for us.. at least not in the "economy" assembly offer.

WROG208, 2023-12-01:
I was wondering if instead of using the connectors that you guys recommend these can be used if they will work?
https://www.amazon.com/gp/product/B08Z7LNTVW/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1
and the same ones but on the other size?

skuep, 2023-12-30:
I think they should work, depending on what radio you want to use.
I will however close this particular issue, since it has drifted away somewhat from the original topic.
Feel free to re-open.

closed 2023-12-30 by skuep

alper061, 2024-04-12:
I would like to reopen it and discuss more.