skuep/AIOC #46: Use the CM108 vendor and product IDs for AllStar Node compatibility.

G1LRO, 2023-11-11:
It's possible to use the AIOC with Allstar Node software if the CM108 emulation could be recognised by the existing software.

The Allstar software identifies the presence of the CM108 by these identites:

#define C108_VENDOR_ID        0x0d8c
#define C108_PRODUCT_ID      0x000c

If the AIOC used these identities, then the Allstar software would work with the AIOC without modification (in VOX mode).

I've worked with David NR9V to make the AIOC work with Allstar by modifying the code to recognise the AIOC, but to save modification of the main branch, using these identities will provide a solution for all current software releases.

There may be other software that could also use the CM108 functionality if it uses the original identities to recognise the chip.

73 Mark G1LRO

skuep, 2023-11-12:
Thanks for all the Allstar Node interest and testing. I don't have a ham license so I cannot really test this stuff on my own.

Regarding the USB VID:PID, let me first say that we cannot just change these values to the ones from CMedia (or any other vendor), because they paid money for their VID block and thus we cannot hijack it. We have our own AIOC VID:PID donated from http://pid.codes which we also use (or are going to use) to find AIOC devices connected to the PC (e.g. the future python config application).

Having said that, I plan to implement configurability of the AIOC (see https://github.com/skuep/AIOC/pull/37) in the future and we could add configurability of the VID:PID so the user can set the VID:PID to whatever they like (on their own responsibility). If that happens to be the CM108 VID:PID so be it ;-) Time-wise I plan to have some holidays at the end of the year dedicated to the AIOC. Besides the configurability, I want to integrate the Virtual-COS feature for general availability. This is a feature specificly designed for Allstar, so the AIOC can send the Allstar software the COS signal by thresholding incoming audio data from the radio. (Basically Virtual-PTT in reverse).

However after changing this VID:PID on the AIOC, it will be harder to run the python config application, because the exact location or VID:PID:serial path will need to be specified. That's definitely a caveat.

davidgsd, 2023-11-12:
If the USB VID:PID was easily configurable that would make it much easier to use the AIOC on any new or existing AllStar node. Otherwise ASL has to modified and compiled (which is easy for software engineers but not easy for average hams), or the ASL code would have to be updated which can take months or years by the time the changes make it into an official release package.

Re. Virtual-COS, ASL does provide this in its usbradio driver carrierfrom=vox mode, however it does not have a configurable activation threshold, and thus while it generally works pretty well, it does require a significant amount of background noise or no long pauses when talking for RX Carrier Detect to remain set. If the activation threshold could be set very low in the AIOC (eg. -60 – -80 dBFS) and also provide a configurable hang time (eg. 100 - 1000 mS) it could work much better than ASL's vox mode.

Also note that Hardware COS is a requirement for most AllStar nodes, as it is the only way to have 100% reliable RX Carrier Detect operation in the majority of personal nodes - which often use lower cost radios, or are radio-less (direct mic input with mic PTT as COS), and thus do not have "flat" discriminator audio available on higher-end radios, which enables ASL's usbradio DSP mode to be used for squelch and CTCSS detection, in which case ASL does not need a COS input. See my How-To guides at https://allscan.info for more detail on AllStar's capabilities and typical configurations in the 3 most common node types (radio-less, HT-based, and mobile / commercial radio / repeater nodes).

Thanks, David NR9V

skuep, 2023-12-30:
I have added the firmware implementation of this feature to the master branch. I have a small hacked-together python script that can be used to set the VID/PID. See here:

````
import hid
from struct import Struct
from enum import IntEnum, IntFlag

class Register(IntEnum):
    USBID = 0x08
    PTT1CTRL = 0x10
    PTT2CTRL = 0x11

class Command(IntFlag):
    DEFAULTS = 0x10
    RECALL = 0x40
    STORE = 0x80

def read(device, address):
    data = device.get_feature_report(int(address), 5)
    address, value = Struct('<BL').unpack(data)
    return value

def write(device, address, value):
    data = Struct('<BL').pack(address, value)
    device.send_feature_report(data)

aioc = hid.Device(vid=0x1209, pid=0x7388)

print(f'Manufacturer: {aioc.manufacturer}')
print(f'Product: {aioc.product}')
print(f'Serial No: {aioc.serial}')

if False:
   print(f'Loading Defaults...')
   write(aioc, 0, Command.DEFAULTS)

if False:
    write(aioc, Register.USBID, (0x1209 << 0) | (0x7388 << 16))
    print(f'Now USBID: {read(aioc, Register.USBID):08x}')

if False:
    print(f'Storing...')
    write(aioc, 0, Command.STORE)
````

I am working on a release of this (and other features) and will provide a binary file, along with a more polished script in the future.

skuep, 2024-01-01:
Find here the pre-release https://github.com/skuep/AIOC/releases/tag/v1.3.0-rc.1

skuep, 2024-03-24:
I think your issue should be solved, so I am going to close the issue for now. If you still have questions or problems, feel free to reopen.

closed 2024-03-24 by skuep