skuep/AIOC #48: Feature Request: Hardware COS for Baofeng UV-5R

rxt1077, 2023-12-01:
I just wanted to mention that there is a simple way to do hardware COS without opening the radio on Baofeng UV-5Rs. There's a [DC bias in the speaker output](https://www.reddit.com/r/amateurradio/comments/rpo031/is_the_uv5rs_36v_dc_bias_on_the_audio_out_typical/) that you can use to trigger a GPIO pin when the squelch is open. [You can also see it used in this YouTube video](https://www.youtube.com/watch?v=QopvHGM-9_k).

I added a diode and resistor to my AIOC and then ran it to a GPIO pin on a Raspberry Pi. I imagine you could do something similar on the AIOC PCB and then return the COS status vi CM108 emulation (or some other way). I tried wiring it up on the schematic but got really stumped trying to fit anything onto that tiny little PCB. It's my first go in KiCad.

Anyway, this is what I was thinking:

[image]

I tried adding this to #17 but wasn't able to re-open the issue.

skuep, 2023-12-30:
Hi, sorry for the late reply. That's an interesting solution that I might add in a future hardware revision. But I think it will only be limited to this one radio, correct?

waydenie, 2024-01-05:
This is an interesting possibility, does the AIOC firmware already support hw COS on any of the input GPIO's? ( even if current devices would need to be modified with a board bodge? )

It might be nice for a future revision of the board and fw to include a pin with current limit/protection brought out to a pad for external COS options. Perhaps like shown in the schematic above , it could have a shorting pad to RADIO-SPK so that you can either connect it there to take advantage of a radio with AF DC bias, or leave it unshorted to connect your own external hw COS hardware.

skuep, 2024-01-05:
Stay tuned for the next hardware revision.. It will be able to use either PTT pin as hardware COS 😁 Might be a nice way to easily hack in this feature as a side effect.. but I need to think about how to split the bias signal from the actual AF signal so it doesn't disturb the COS detection...

W1BTR, 2024-07-14:
I've been working on the AIOB, and a big request I get is a hardware COS input and output pad. Is there a pin on the STM chip that is already ready for a hardware COS input?

skuep, 2024-07-14:
There are long-standing future plans on 2 additional inputs on the STM32 that could be used for COS (eventually). Check out this branch: https://github.com/skuep/AIOC/tree/k1-aioc-rev1.1

For reference, here the schematic:
[image]

PB6 and PB7 are two inputs that are connected to the PTT1/2 pins in hardware, so we get combined GPIOs. However in software, the inputs and outputs will be handled individually so that you can easily use PB6/PB7 as inputs for this. They are also 5V capable btw.

W1BTR, 2024-07-15:
> There are long-standing future plans on 2 additional inputs on the STM32 that could be used for COS (eventually). Check out this branch: https://github.com/skuep/AIOC/tree/k1-aioc-rev1.1
> 
> For reference, here the schematic: ![Screenshot from 2024-07-14 16-08-58](https://private-user-images.githubusercontent.com/5630293/348548704-fbf4dcc6-cc01-4a9c-b30b-ab235121d137.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjEwMTM0OTUsIm5iZiI6MTcyMTAxMzE5NSwicGF0aCI6Ii81NjMwMjkzLzM0ODU0ODcwNC1mYmY0ZGNjNi1jYzAxLTRhOWMtYjMwYi1hYjIzNTEyMWQxMzcucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcxNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MTVUMDMxMzE1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NDI0NGZkNTA2YjQ0NGU5OTA0ZTI3MzA1MTIxODM4MDZiMjY2MTM4M2RlOGI2N2ZhNDhiMWEyNTgxOTcwOGY3ZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.0BpOdiQq_GHCr6kQ6PpkCzH69AxdqixrmeTlW71h0kQ)
> 
> PB6 and PB7 are two inputs that are connected to the PTT1/2 pins on the AIOC board, so we get combined GPIOs. However in software, the inputs and outputs will be handled separately so that you can easily use PB6/PB7 as inputs for COS. They are also 5V capable btw. I suggest we could use the pin labeled PTT2-IN to default as COS input.

Ah, that's where the split pad is. That's great! Ill have to compare the changes to make sure I can update the AIOB

G1LRO, 2024-08-20:
Hi Simon, I'll give this thread a bump up.

I've now been making radio implementations using PB7 active low for hardware COS, so hopefully the code can be grafted into V1.3 (keeping the support for CM108 IDs etc.) 

I'll encourage others in the maker community using the AIOC CPU to contribute to this thread also.

skuep, 2024-09-17:
I have had a quick go at it, see my branch here:
https://github.com/skuep/AIOC/tree/hwcos

Here is the compiled firmware where virtual COS is disabled and instead PB7 (PTT2) pin is routed to CM108 volume down and serial DCD pin. The PB7 pin is low-active (i.e. has to be pulled to ground) in order for the "volume down" and DCD signal to be activated.

[aioc-fw-1.3.0-rc.1-hwcos.zip](https://github.com/user-attachments/files/17028211/aioc-fw-1.3.0-rc.1-hwcos.zip)

Regarding the new hardware revision, how much interest is there in it? I have a few of them here, but I keep delaying the "release" in fear of not testing it well (I don't have the means to test it properly). For a release of the new hardware (with split pad and hardware input support) I need some testers... Let me know if you are interested.

skuep, 2024-09-18:
Official v1.3.0-rc.2 is now live for testing
https://github.com/skuep/AIOC/releases/tag/v1.3.0-rc.2

skuep, 2025-01-04:
Work is ongoing on the next public hardware release (will be v1.2) with many new features, hardware COS on PTT1/PTT2 being one of them. Firmware support is there already for hardware-COS since v1.3.0-rc.2 as mentioned above.

I am closing this issue now. Feel free to reopen if there are still open questions or problems.

closed 2025-01-04 by skuep

rxt1077, 2025-01-04:
Thank you for working on this! I haven't had a chance to do another hardware run, but when I do I'll be sure to check it out.