skuep/AIOC #49: Unexpected error when compiling

TonyWBI, 2023-12-17:
Complete newbie to STM32CubeIDE here. Saw the AIOC project and thought I would give it a go. 
I think I've imported the project software OK, but I get this error message.
```

_16:59:39 **** Build of configuration Debug for project aioc-fw ****
make all 
arm-none-eabi-gcc "../Src/usb.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DSTM32F302xC -DCFG_TUSB_MCU=OPT_MCU_STM32F3 -c -I../Drivers/CMSIS/Include -I../Drivers/CMSIS/Device/ST/STM32F3xx/Include -I../Drivers/STM32F3xx_HAL_Driver/Inc -I../Src -I../Inc -I../Middlewares/Third-Party/tinyusb/src -Og -ffunction-sections -fdata-sections -Wall -fstack-usage -fcyclomatic-complexity -MMD -MP -MF"Src/usb.d" -MT"Src/usb.o"  -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Src/usb.o"
../Src/usb.c:4:10: fatal error: tusb.h: No such file or directory
    4 | #include "tusb.h"
      |          ^~~~~~~~
compilation terminated.
make: *** [Src/subdir.mk:50: Src/usb.o] Error 1
"make all" terminated with exit code 2. Build might be incomplete.

16:59:40 Build Failed. 2 errors, 0 warnings. (took 1s.324ms)_
```

There isn't (or doesn't seem to be) a tusb.h anywhere in the project, so I'm guessing it's a 'built in'  file.

I've tried the complile on a Linux and a Windows PC - same error.  Mr. Google had a reference to a missing tusb.h file, but in relation to compiling on PlatformIO for the ESP32.

Is it a stupid newbie error??

TIA,
Tony...

skuep, 2023-12-17:
Did you run the submodule update command?

https://github.com/skuep/AIOC?tab=readme-ov-file#how-to-build

TonyWBI, 2023-12-17:
Numpty Error in Line 1 : failed to follow the instructions. :-(

In my defence, I had done the submodule update the first time I downloaded the source, but having failed a few times to import the code into STMCubeIDE, I ended up downloading it a couple more times, but didn't do the update. Didn't do it on the Windows PC either.

So I have a successful build. Yippee!! Another newbie question - is it the resulting elf file I need now? Last time I programmed an STM32, I had a hex file, but that was using an ST-Link programmer.

skuep, 2023-12-18:
Nice!
I think you need the bin file (or hex file, not sure). By the way, there are also compiled binary files available in the GitHub releases section.

TonyWBI, 2023-12-18:
Thanks for all of your help and patience, but I'm putting this exercise on the 'Too Difficult' pile.
When compiling, all I get is the elf file. To get a bin and/or hex file, I need to change some project properties, but the the properties are greyed out.
Also, dfu-util says 'No DFU capable USB device available', so I've got something going wrong there as well.

skuep, 2023-12-19:
Actually, it should convert the elf file to a bin file in the post build process. Are there any errors in the log?

It's the device unprogrammed? Then it needs to be set into dfu mode first using the boot pin. It's all in the readme file ;-)

TonyWBI, 2023-12-19:
This is the end of the console log : 
```
12:48:17 **** Incremental Build of configuration Debug for project aioc-fw ****
make all 
arm-none-eabi-size  aioc-fw.elf 
   text	   data	    bss	    dec	    hex	filename
  46552	   1908	   4740	  53200	   cfd0	aioc-fw.elf
Finished building: default.size.stdout
12:48:18 Build Finished. 0 errors, 0 warnings. (took 276ms)
```
It's already been built once, which it was only 276mS this time, but it was the same ending last time.

However, I've just done another search and found this - https://community.st.com/t5/stm32cubeide-mcus/how-to-generate-bin-file-after-building-with-cubeide/td-p/334423 - and I now have a bin file. Whether it is right or not I don't know. 

As to the programming, I plugged in the board to USB with the two pins shorted. `lsusb` recognises the device - `Bus 003 Device 032: ID 0483:df11 STMicroelectronics STM Device in DFU Mode` .  
However - `dfu-util -d  0483:df11 -a 0 -s 0x08000000:leave -D aioc-fw.bin` (I changed the ID numbers to my device) returned 
```
dfu-util 0.11-dev

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release
dfu-util: Cannot open DFU device 0483:df11 found on devnum 37 (LIBUSB_ERROR_ACCESS)
dfu-util: No DFU capable USB device available
```
Something to do with LIBUSB??
So one step forward, but I'm on the limits of my knowledge here, hence the struggle.

skuep, 2023-12-19:
Regarding first issue: can you switch the configuration to Release (instead of Debug)

Second issue: try running dfu-util as root (using sudo).

TonyWBI, 2023-12-21:
I really should know better. Anything with Linux - if it doesn't work as the user, try it with sudo! Minus several million geek points for me. :-)
I have now flashed 5 devices with the bin file I created earlier, and each device appears as ttyACM0 and as a sound card. I now have to connect the to a radio and see if they work.
UPDATE - Reading and writing to a couple of different Baofeng radios with CHIRP has worked. Also it is receiving packets with Direwolf, but I'm trying to get the PTT settings right. I'm reading through the https://github.com/wb2osz/direwolf/issues/448 issue now.

Once again, Simon - thank you for your patience and assistance.

skuep, 2023-12-30:
You're very welcome! 
You should definitely also try ``PTT CM108`. It should be a much neater solution and with the most current Direwolf version there won't be any hiccups. However on Linux you might need to set some permissions for the ``hidraw`` device, but Direwolf will tell you.

I am going to close this issue. Feel free to re-open if you have any more issues.

closed 2023-12-30 by skuep