skuep/AIOC #50: No activity, no DFU mode

klinquist, 2024-01-08:
Hi, 
Thank you for this project! I followed your directions and got my package of 5 quickly from JLCPCB.

However, it's not entering DFU mode.  Any troubleshooting I can do? Measure voltage(s) somewhere on the board?   Same results with all of them...

```
$ dfu-util -a 0 -s 0x08000000 -D aioc-fw-1.2.0.bin
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Match vendor ID from file: 1209
Match product ID from file: 7388
dfu-util: No DFU capable USB device available
```

[image]

desaiamogh, 2024-01-09:
@klinquist Do you have any of those extra boards ? Can you spare one?

skuep, 2024-01-09:
That's the wrong pins that are shorted. Needs to be the outer most pins. See the silkscreen on the back.

Also, you should try adding ``sudo`` before the dfu command.

klinquist, 2024-01-09:
ah-ha!  Thank you @skuep , got all 4 flashed - now waiting for the 3.5mm and 2.5mm jacks to be delivered from Adafruit.   

@desaiamogh Not currently, but I just followed skuep's directions and had 5 of them in my hand in about 5 days for $65 USD!

skuep, 2024-01-09:
Nice! 😎 Thanks a bunch for the pull request. Much appreciated!

PS: of you order ten of them, they are even cheaper per piece 😁