skuep/AIOC #51: Integrate a TNC Modem with KISS interface

desaiamogh, 2024-01-09:
Hi,  In your future work section you mentioned about "Maybe integrate a TNC Modem with KISS interface? (I am not sure if that is worth the effort)"

It would help a great deal to the community if a KISS tnc which can do 1200/9600 baud can be added to this cable.

Most 9600 baud tnc projects use rare components and there are hardly any designs currently available in production or sale which were released few years ago. I believe almost all projects died in the covid aftermath. There are so many email over VHF projects like TARPN that are alive and popular, but new members joining these networks is very difficult due to unavailability of TNCs, modems and need of specific ancient data rigs, etc.

Is there a plan for integrating a kiss modem?

desaiamogh, 2024-01-09:
Similar work seems to have been done at
https://github.com/mobilinkd/NucleoTNC

using a STM32L432KC

skuep, 2024-01-20:
Sorry for the late reply. Yeah that sounds certainly interesting. Although at the current point I would strongly suggest to separate this into a separate firmware, since it's not easy at all to integrate this into the current firmware. 

Unfortunately I don't have enough time on my hands for this, nor do I have the right experience, equipment and also no HAM license. But I am open to having people work on porting the NucleoTNC project to the AIOC hardware. It should be compatible with their license, from what I can tell.

zapewne, 2024-05-22:
I need any way to use reticulum https://reticulum.network/manual/interfaces.html
any interface is ok.

W1BTR, 2024-07-20:
> I need any way to use reticulum https://reticulum.network/manual/interfaces.html any interface is ok.

In the meantime, Direwolf runs a KISS server with AX.25, so that could be used with the AIOC. Itll run on linux windows, etc. A pizero 2 w would work well - a standard pi zero w might be a bit weak. Any full raspberry pi would also work

skuep, 2025-01-04:
For documentation purposes I wanted to leave a link to this GitHub repo, which was recommended on Discord by VE7TMX
https://github.com/sq8vps/vp-digi

This looks like a cool solution/alternate firmware that could potentially be ported to the AIOC

W1BTR, 2025-01-09:
I dont use terms like this very often as Im not really a gen z (age wise I guess Im on the line) but...

*clears throat*

_"hype."_

skuep, 2025-01-11:
> I dont use terms like this very often as Im not really a gen z (age wise I guess Im on the line) but... um... *clears throat* **hype!**

I assume that this is a positive emotion towards the idea... ;-) ;-) I am in contact with Piotr from VP-Digi and I will send him a fresh AIOC v1.2 hardware. He wanted to look into the idea of porting his TNC software to the AIOC.

W1BTR, 2025-01-11:
Definitely a positive reaction, yes.

desaiamogh, 2025-01-12:
> I assume that this is a positive emotion towards the idea... ;-) ;-) I am in contact with Piotr from VP-Digi and I will send him a fresh AIOC v1.2 hardware. He wanted to look into the idea of [...]

Just an year into my initial request, seeing such an update makes me feel very +ve about this (at least considering usual slow progress in the hobby and opensource area)

Thanks for the update.

skuep, 2025-09-26:
I am not sure, I think I totally forgot to post an update in this issue here. Piotr did successfully port the VP-Digi firmware to the AIOC. See his branch here https://github.com/sq8vps/vp-digi/tree/aioc.

He also sent me a hex file, which I converted into a bin file (see attachment). I tested it, and it enumerates sucessfully as a USB device. The firmware file can be flashed with the usual way (see README). Note however, if you want to change back to the AIOC firmware, the "Initial programming" needs to be followed (with the short jumper cable).

[vp-digi-aioc.bin.zip](https://github.com/user-attachments/files/22563470/vp-digi-aioc.bin.zip)