skuep/AIOC #54: PTT Remains engaged and Audio Device crashes on Linux and Mac

b4ux1t3, 2024-01-20:
Whenever I transmit using a UV-5R, the AIOC seems to crash, making the audio and serial devices disappear and sticking the PTT on permanently.

Unplugging and plugging it back in works just fine, but I can't transmit at all.

This happens in both classic serial PTT and CM108 PTT modes.

Direwolf throws this at me:

```
ioctl HIDIOCGRAWINFO failed for /dev/hidraw0. errno = 0.
Audio input device 0 error code -19: No such device
...
Audio input device 0 error code -19: No such device
Terminating after audio input failure.
```

I haven't tested on windows yet, but I have tested with a Mac, a Pi, and a linux laptop.

skuep, 2024-01-20:
Can you have a look at this issue?
https://github.com/skuep/AIOC/issues/11

There have been some problems in the past regarding with EMI issues. Maybe you can find a workaround there

b4ux1t3, 2024-01-20:
Ah, I gave it a glance and thought it didn't have anything to do with me, but reading closer there are some things I haven't tried yet.

I don't have the same issues on Windows (completely different issues, haha), but it's similar enough that it's worth trying. Thanks!

I'll give some of it a try and update.

skuep, 2024-01-20:
Ah sorry, you are right. I should close that one.
I meant to say this issue:
https://github.com/skuep/AIOC/issues/22

skuep, 2024-03-24:
I assume your problem is solved or you found another solution and I am going to close this issue. Feel free to reopen if you need more help.

closed 2024-03-24 by skuep

b4ux1t3, 2024-03-28:
I just haven't had any free time to look at it. Ha! I'll get to it eventually, and reopen if necessary, thanks!