skuep/AIOC #55: create a group buy list

sysadminpower2019, 2024-01-30:
Hello,

so i'd love to support the project by buying directly from the maker but shipping to the european continent plus import taxes makes it incredibly expensive.
Maybe instead we could set up a list where interested people in different parts of the world can act as points of contact and buy for all the locally interested people

skuep, 2024-02-04:
Thanks for your interest. Just for the record, I don't sell these myself (but I am based in Germany).
The best bet is to get 5pcs made by JLCPCB, I think it is around 120€ or so, and you have 4 others to sell/give away.

I think this platform is not ideal for a group buy list, since people usually only come here when they have issues. I suggest you can visit the Discord channel (linked on the GitHub page). Unfortunately, it is mostly populated with US people.

maggo1404, 2024-02-11:
Which country do you live in?

skuep, 2024-03-24:
Due to inactivity I am going to close this issue. Feel free to reopen if required.

closed 2024-03-24 by skuep