skuep/AIOC #56: Unable to flash Firmware with AIOC

magellan-13016, 2024-02-04:
Hello,

First many thanks for this very useful device ! 

All is working perfectly : Chirp reading and writing, audio card... except firmware flashing.

I'm using google Chrome to flash with this link (working with official programming cable) : 
[](https://egzumer.github.io/uvtools/)

But with AIOC, It does nopt work. The browser return this :

**CRC check passed...
Detected firmware version: *EGZUMER 9816c1
Firmware uses 96.33% of available memory (59184/61439 bytes).
Connecting to the serial port...
No data received, is the radio connected and in flash mode? Please try again.**

Thanks for working on it if you can and have time for.

David (F4BPP).

pfeerick, 2024-02-08:
I had the same problem with a board I just had made with JLCPCB (running 1.2.0 firmware), using the uvtools website and Brave web browser on Ubuntu Linux (with web serial enabled). It would fail at random times all throughout the download from 2% to 85%, so I swapped to a dedicated cable I had for the baofengs and it was fine. 

FWIW, I noticed the blue light came on with the dedicated cable, but not with the AIOC adapter.

skuep, 2024-02-09:
Interesting.. I have a few suspects on what the problem could be.. I think I will put out some testing firmware for you to narrow down the problem and fix it for good

skuep, 2024-02-11:
I switched course and I have bought a Quansheng UV-K5 myself, so I can debug this much more easily :-) They are so cheap..

3DFabrik, 2024-02-11:
Hi Simon,
We have already chatted via email and you asked me to write down my experience with your adapter, so here goes.
I am having the same problems with flashing my Quansheng, both via the browser and the Quansheng flash software that comes with it. The update stops randomly and it looks like there might be a timing issue.
Also, the Adpater seems to reset when I try to PTT via the Quansheng Dock software. However, with the Quansheng Dock the serial link works fine and I can read the memory and send commands to the radio.
Thank you for this great project and all your efforts. I know how much work it is.
Best regards
Patric

skuep, 2024-02-18:
Thanks for the report Patric! I just programmed the egzumer firmware to my new Quansheng UV K5-8 using the browser flasher without any issues.. I am running on Linux so I am currently thinking that it must be an OS problem.

Do (all of) you guys use Windows per chance? In order to narrow down the issue, I have attached a few test firmwares that you can try. Let me know if you have success with one of the firmwares. I tried changing a few settings and also increasing serial buffer size.
[aioc-fw-1.3.0-rc.1-noserialptt.zip](https://github.com/skuep/AIOC/files/14323491/aioc-fw-1.3.0-rc.1-noserialptt.zip) (PTT from RTS/DTR signals is disabled)
[aioc-fw-1.3.0-rc.1-rxignptt-none.zip](https://github.com/skuep/AIOC/files/14323492/aioc-fw-1.3.0-rc.1-rxignptt-none.zip) (RX ignore when PTT asserted disabled)
[aioc-fw-1.3.0-rc.1-serialbuf.zip](https://github.com/skuep/AIOC/files/14323493/aioc-fw-1.3.0-rc.1-serialbuf.zip) (Serial buffer size doubled)

magellan-13016, 2024-02-21:
Thank you for your investigations but the initial problem for me was not an interrupted flash but a flash impossible to start because the Chrome browser under Kubuntu 22.04 systematically returns this error:

> CRC check passed...
> Detected firmware version: *EGZUMER 9816c1
> Firmware uses 96.33% of available memory (59184/61439 bytes).
> Connecting to the serial port...
> No data received, is the radio connected and in flash mode? Please try again.

So should I try one of the three firmware to solve my problem, and if so, which one?

skuep, 2024-02-21:
Did you put the radio in firmware flashing mode (PTT key during power up)?
Are you on Windows or Linux?

magellan-13016, 2024-02-21:
Of course I  put the radio in firmware flashing mode. I'm flashing my Quansheng with a new firmware every week at this time (the one of F4HWN based on egzumer, updated once a week) . I have no problem with the original flashing cable but only with AIOC. That's not a problem for me since I have both but I would like to talk about AIOC on my website and tell people that they don't have to buy the original cable to flash the quansheng and buy an AIOC instead so they can do all with it.

skuep, 2024-02-21:
Can you try the group thing? The AIOC is a CDC device,  which can by default only be accessed by the root user.

pfeerick, 2024-02-22:
Do a `ls -la /dev/ttyUSB*' with the AIOC connected, and see what group it is assigned to. If like me it was 'dialout', then check with `groups` if your username is in that group also. If not, do a `sudo usermod -aG dialout your_username` to add your user to the dialout group, and then either logout and login again, or simply restart the computer, to apply the group changes.

skuep, 2024-02-22:
> I don't think it matters and don't know how to manage that.

Given that I had the same issues, I think it matters a lot. Thanks @pfeerick for the instructions

> Do a `ls -la /dev/ttyUSB*` with the AIOC connected

It should be `ls -la /dev/ttyACM*` but other than that youre correct! You can also go to the User/Groups Editor of the operating system, select your user and add your user to the group 'dialout'.

If you don't want to, you can alternatively create a udev rule for the AIOC device that allows anyone on the system to use it. Put ``SUBSYSTEM=="usb", ATTR{idVendor}=="1209", ATTR{idProduct}=="7388", MODE="0666"``  in a file ``90-AIOC.rules`` under directory ``/etc/udev/rules.d``.

See here for more information: 
https://askubuntu.com/questions/15570/configure-udev-to-change-permissions-on-usb-hid-device

ErikS-web, 2024-05-26:
Finally I got my an AIOC and start testing ...

In am using latest [**v1.3.0-rc.1**](https://github.com/skuep/AIOC/releases/tag/v1.3.0-rc.1)
The interface works ok on **Q-dock** and **Chirp**. Both on **_Windows-11._**

Now testing [Online flash-tool of Egzumer](https://egzumer.github.io/uvtools/?firmwareURL=https://github.com/egzumer/uv-k5-firmware-custom/releases/download/v0.22/egzumer_v0.22.packed.bin) with using **Windows-11** and **Google Chrome.**.

[image]

But when **Flashing is NO Suc6,** 
I did some retries with increasing Volume. (Had to ReBoot UVK5 every time)

> Loading file from url: https://github.com/egzumer/uv-k5-firmware-custom/releases/download/v0.22/egzumer_v0.22.packed.bin
> 
> CRC check passed...
> Detected firmware version: *EGZUMER v0.22
> Firmware uses 98.19% of available memory (60328/61439 bytes).
> Connecting to the serial port...
> No data received, is the radio connected and in flash mode? Please try again.
> Connecting to the serial port...
> Radio in flash mode detected.
> Bootloader version: 2.00.06
> Version check passed.
> Flashing firmware...
> Flashing... 3.0%
> Flash command rejected. Aborting.
> No data received, is the radio connected and in flash mode? Please try again.
> Connecting to the serial port...
> Radio in flash mode detected.
> Bootloader version: 2.00.06
> Version check passed.
> Flashing firmware...
> Flashing... 7.6%
> Flash command rejected. Aborting.
> No data received, is the radio connected and in flash mode? Please try again.
> Connecting to the serial port...
> Radio in flash mode detected.
> Bootloader version: 2.00.06
> Version check passed.
> Flashing firmware...
> Flashing... 3.0%
> Flash command rejected. Aborting.
> No data received, is the radio connected and in flash mode? Please try again.

Flashed radio now with CH340 interface and UVtools correctly and than next test ...

Also **No suc6 with K5Prog-Win V1.27**

> Loaded 60346 (0xEBBA) bytes (max 0xF000) from ..
> D:\egzumer_v0.22.packed.bin
> 
> firmware file de-obfuscated
> 
> firmware file version '*EGZUMER v0.22'
> 
> 'COM4' opened at 38400 Baud
> 
> Uploading firmware to the radio ..
> 
> listening for firmware update mode packet ..
> 
> Bootloader version '2.00.06'
> 
> 
> write_flash  offset 0000  len 256
> write_flash  offset 0100  len 256
> write_flash  offset 0200  len 256
> write_flash  offset 0300  len 256
> write_flash  offset 0400  len 256
> write_flash  offset 0500  len 256
> write_flash  offset 0600  len 256
> write_flash  offset 0700  len 256
> write_flash  offset 0800  len 256
> write_flash  offset 0900  len 256
> write_flash  offset 0A00  len 256
> write_flash  offset 0B00  len 256
> write_flash  offset 0C00  len 256
> write_flash  offset 0D00  len 256
> error: k5_write_flash() [0]
> 
> 'COM4' closed
> 'COM4' opened at 38400 Baud
>

skuep, 2024-05-26:
Yeah, can you try the firmware aioc-fw-1.3.0-rc.1-serialbuf.zip from above?

ErikS-web, 2024-05-26:
As you can see above... I works a litte bit and than fault.
So I decided to test [aioc-fw-1.3.0-rc.1-serialbuf.zip](https://github.com/skuep/AIOC/files/14323493/aioc-fw-1.3.0-rc.1-serialbuf.zip) (Serial buffer size doubled)

As the AIOC is already programmed, I do not need the wire and can use the command ...
`dfu-util -d 1209:7388 -a 0 -s 0x08000000:leave -D aioc-fw-1.3.0-rc.1-serialbuf.bin`

I got an error

> Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.

> Copyright 2010-2021 Tormod Volden and Stefan Schmidt
> This program is Free Software and has ABSOLUTELY NO WARRANTY
> Please report bugs to http://sourceforge.net/p/dfu-util/tickets/
> 
> Opening DFU capable USB device...
> Device ID 1209:7388
> Run-Time device DFU version 0101
> Claiming USB DFU (Run-Time) Interface...
> **Cannot claim interface 6: LIBUSB_ERROR_NOT_SUPPORTED**

[Here ](https://stackoverflow.com/questions/72068206/libusb-open-returns-libusb-error-not-supported-on-windows-10) I found the tip, to use [ZADIG ](https://zadig.akeo.ie/) again and install **Libusb-WinUSB**

[image]

And now `Programming without the Wire` works with command

`dfu-util -d 1209:7388 -a 0 -s 0x08000000:leave -D aioc-fw-1.3.0-rc.1-serialbuf.bin`

> dfu-util 0.11
> 
> Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
> Copyright 2010-2021 Tormod Volden and Stefan Schmidt
> This program is Free Software and has ABSOLUTELY NO WARRANTY
> Please report bugs to http://sourceforge.net/p/dfu-util/tickets/
> 
> Opening DFU capable USB device...
> Device ID 1209:7388
> Run-Time device DFU version 0101
> Claiming USB DFU (Run-Time) Interface...
> Setting Alternate Interface zero...
> Determining device status...
> DFU state(0) = appIDLE, status(0) = No error condition is present
> Device really in Run-Time Mode, send DFU detach request...
> Device will detach and reattach...
> Opening DFU USB Device...
> Claiming USB DFU Interface...
> Setting Alternate Interface #0 ...
> Determining device status...
> DFU state(10) = dfuERROR, status(10) = Device's firmware is corrupt. It cannot return to run-time (non-DFU) operations
> Clearing status
> Determining device status...
> DFU state(2) = dfuIDLE, status(0) = No error condition is present
> DFU mode device DFU version 011a
> Device returned transfer size 2048
> DfuSe interface name: "Internal Flash  "
> Downloading element to address = 0x08000000, size = 128000
> Erase           [=========================] 100%       128000 bytes
> Erase    done.
> Download        [=========================] 100%       128000 bytes
> Download done.
> File downloaded successfully
> Submitting leave request...
> Transitioning to dfuMANIFEST state

So now I can test loading firmware agaian....

My findings .... Indeed now it works BUT **only with little volume** 

Just switch on radio and `do not use more volume than 9 o'clock`
Than is works fine.
[... 78 lines cut ...]
> Radio in flash mode detected.
> Bootloader version: 2.00.06
> Version check passed.
> Flashing firmware...
> **Flashing... 66.2%**
> **Flash command rejected. Aborting.**
> No data received, is the radio connected and in flash mode? Please try again.
> Connecting to the serial port...
> Radio in flash mode detected.
> Bootloader version: 2.00.06
> Version check passed.
> Flashing firmware...
> **Flashing... 4.2%**
> **Flash command rejected. Aborting.**
> No data received, is the radio connected and in flash mode? Please try again.
> Connecting to the serial port...
> Radio in flash mode detected.
> Bootloader version: 2.00.06
> Version check passed.
> Flashing firmware...
> **Flashing... 100%**
> **Successfully flashed firmware**
> 

So **_bigger buffer works, but programming is very sensitive to volume of radio_**

skuep, 2024-05-26:
Thanks a lot for the detailed testing! Yeah, windows is very picky with the DFU util stuff. For some people it just works, others have to use weird tricks. I don't know why it's all over the place. Unless someone can tell me a specific problem, I am leaving it as is, because there is not much that is more painful than debugging Windows issues and also because I don't have any Windows machine anymore at home. 

The volume thing is odd. Can you or anyone confirm that this is only a problem with the AIOC and not with an official cable? What if you have the volume at minimum?

skuep, 2024-09-18:
v1.3.0-rc.2 now has the doubled buffer size fix included.
https://github.com/skuep/AIOC/releases/tag/v1.3.0-rc.2

I will close this issue for now, just reopen if you need more help.

cdidier, 2025-05-05:
Had to bump the buffer from 256 to 512 to be able to flash on my setup (debian testing, flashing through google chrome). Flashing would fail randomly so I've just doubled the CDC serial buffer size again to 512 (like in https://github.com/skuep/AIOC/commit/e93be6926f98411527e16a1191563238e23a1882 )and now it seems to work flawlessly.

skuep, 2025-05-05:
Thanks a bunch for the info. I will double the buffer again in the next firmware.

skuep, 2025-09-26:
Solved in v1.4.0

closed 2025-09-26 by skuep