skuep/AIOC #57: Parts unavailable at jlcpcb

davistroy, 2024-02-05:
the ldo and a green led are both unavailable at jlcpcb ... i can substitute the green led easily, but any thoughts on a substitute for the ldo without re-spinning the board?

jakern, 2024-02-06:
Commenting to follow along. I'm probably just  going to have to wait. Adding 'out of stock' for the next person.

skuep, 2024-02-06:
You can have JLCPCB buy the parts at Digi key (among others) and mount them for you. Look into your JLCPCB profile, there is a tab about part sourcing. You can have them order a specific part at your distributor and then you wait for the part to arrive in China. Then your can put your order and select your own personal stock for the out-of-stock part

skuep, 2024-03-24:
Due to inactivity I am going to close this issue. Feel free to reopen if required.

closed 2024-03-24 by skuep

W1BTR, 2025-04-19:
@skuep will the board work fine with the green led missing? Not available from any manufacturers and trying to get shipments in before my cost 3Xs from tarrifs.

Radio-X24, 2025-04-19:
> led missing? LED-DIP2-2.54-W is not available from any manufacturers and

It's just an LED I would just select another compatible part if I was you
it won't be hard to match.

LEDs are good to know what's going on

W1BTR, 2025-04-19:
Im not sure what else would be compatible is the issue. The JLCPCB search tool doesnt tell much for someone who doesnt know PCB sizing and such

skuep, 2025-04-20:
> [@skuep](https://github.com/skuep) will the board work fine with the green led missing? LED-DIP2-2.54-W is not available from any manufacturers and trying to get shipments in before my cost 3Xs from tarrifs.

The board does work without but it's nice to have them. What about the part number you mentioned? Sounds wrong, it should be a 0603 led, not 2.54mm DIP. I am currently not at home to see, but it shouldn't be that difficult to find another one.

W1BTR, 2025-04-20:
Was able to find an emerald one. Barely enough in stock, everyrhing else was yellow or red.

Not sure on the part number but I know it worked in the past.