skuep/AIOC #58: Feature Request: Yeasu Style Mic and speaker

fanlessfan, 2024-02-10:
Hello, this is great project. Could you add Yeasu style mic and speaker support? here is the link to it. https://pinoutguide.com/PortableDevices/yaesu_vertex_microphone_pinout.shtml

skuep, 2024-02-18:
From what I can tell it functions quite differently. Not sure if it's adaptable that easily..

fanlessfan, 2024-02-19:
@skuep actually it's very close to the 2 pin Kenwood connector which has Ground, MIC,SPK,PTT and RX/TX. the Pin 2 (Clone) in the diagram is TX/RX which need a diode to split it to two pins (smart). And Yeasu has package TNC cable CT-91 which convert the one pin to two pins. I built a programming cable using the diagram below with diode and it works perfect.
[image]
CT-91
[image]