skuep/AIOC #59: AIOC board plus Bluetooth = HAM radio connected to Vehicle Mic & Speakers?

IoTPlay, 2024-02-11:
Hi, sorry for Logging an issue, this is me just thinking loudly - I did not see a community forum where I could share this idea - point me to one - and close the issue if you do not want this as an issue please.

Could the AIOC board one day be combined with a bluetooth board, and some programming - to let a HAM mobile (like Baofeng) look like a smart cellphone - plus maybe a PTT that is bluetooth operated - to hook up to a car's bluetooth, like your cellphone would?

skuep, 2024-02-18:
It's an interesting idea, but that certainly adds quite a lot of complexity, which I am not ready to maintain (both in software as in hardware). Probably would need to use an ESP32 or something, which I am also not familiar with. Would probably also need to source a battery for the thing. In addition to that, Bluetooth is very, very complex from a software side.

So yeah, cool idea but I currently don't think it would happen, at least not from my side.

skuep, 2024-03-24:
In conclusion, thanks for the suggestion, but it doesn't seem to fit in my time constraints. If anyone else wants to have a go at this, feel free to! So for now I am going to close the issue.

closed 2024-03-24 by skuep

IoTPlay, 2024-05-05:
Excellent, thank you, and thank you for your work here !

goopaga, 2024-05-05:
> Hi, sorry for Logging an issue, this is me just thinking loudly - I did not see a community forum where I could share this idea - point me to one - and close the issue if you do not want this as an [...]

Hi, there's a new great project involving an ESP32 as a radio interface.
See https://github.com/kamilsss655/ESPRI

It's in an early stage but with great milestones achieved so far (management web server, audio input, audio output, CW, FSK and audio beacon, and growing...).

Sorry to hijack this topic but since you were asking for Bluetooth capability this project may achieve that feature, especially if someone can jump in for help in the development.

Cheers

skuep, 2024-08-05:
Sorry for the late reply, just stumbled upon it. Looks like a cool project! Thanks for the link