skuep/AIOC #60: cannot write a bin file in non-BOOT mode

Flying-Pit, 2024-03-01:
Hello
I was able to write the 1.2.0 FW to my AIOC in bootloader mode. After restarting without a boot bridge, the audio device and the serial interface appear in the WIN10 device manager. Only under "Other devices" the AIOC DFU Runtime has a yellow triangle.
If I now try to write to the AIOC with FW 1.3.0-RC1.bin without a boot bridge, the following error occurs.
What can this be?

[image]
[image]

StrongJohn, 2024-03-01:
I believe your issue is that you need to install libusb to AIOC DFU Runtime (Interface 6) using zadig.

Flying-Pit, 2024-03-02:
exactly, that was the solution. thanks friend.

skuep, 2024-03-02:
Windows is such a hit & miss, to be honest. Most people don't have any issues whatsoever and suddenly some systems require explicit drivers that even need to be installed using the Zadig tool. I have no explanation on why things vary so much on Windows OSs. 

Anyway, glad it was solved for you. Btw, the libusb thing on Windows is mentioned in the README. But I did not know that there are so many variants out there. Glad its working now! And yeah, you need dfu-util to talk to both interfaces (Runtime and actual Bootloader). The DFU runtime belongs to the AIOC firmware actually and it's sole purpose is that dfu-util can request the AIOC to go into the bootloader mode, which it re-enumerates in as an STM32 device.

Flying-Pit, 2024-03-02:
Yes, that's my experience too. I work with many bootloaders, only with some I had to explicitly use Zadig and help out. But it works now and the AIOC can be reprogrammed perfectly via USB and DFU-UTIL. Now it's time to test the various handheld radios. 
THANK YOU again for your great work!!!!

skuep, 2024-03-24:
Glad to see it's working for you now! I am going to close this issue, feel free to reopen.

closed 2024-03-24 by skuep