skuep/AIOC #61: AIOC and AllStarLink with simple_usb Module

Murf, 2024-03-02:
More of a question than an Issue,
I am trying to get my AIOC working with a 'feng UV-5R for an AllStarLink node.
I recompiled the chan_simpleusb module with the correct vendor and product ID's, and was surprised when the PTT came on during the simpleusb-tune-menu utility option to flash the PTT.
I know I have a long way to go, especially with COR.
How is COR currently handled by default? is it a code sent on the serial port?
Thanks,
Murf.

Murf, 2024-03-02:
Oh wow I just found the solution already, I was on the right track.
https://www.youtube.com/watch?v=dMoZEgm2ULo&ab_channel=chcmedia FYI

skuep, 2024-03-02:
Nice! So this is a first confirmed use for All-star link for me 🙂 Did you end up using the Virtual COS function of the AIOC?

skuep, 2024-03-02:
Virtual COS essentially listens to the audio from the radio and tries to determine if there is a carrier (squelch disengaged). Currently this is done using very simple thresholding.
It asserts the volume down (iirc) key of the cm108 emulation, no serial codes required.

Murf, 2024-03-02:
Ah yes I saw that option in the ASL module code, cool I will see if I can get it to work!

Murf, 2024-03-02:
Oh.. I think I should be using v 1.3 RC1 not 1.2, correct Simon?

skuep, 2024-03-02:
Ooops, you are correct! My bad. It looks like the COS feature was introduced in the v1.3 version.

Shopstream204, 2024-03-02:
Murf, is it possible to share your modified version of Allstarlink?  I would love to test it with my AOIC.

Murf, 2024-03-02:
All I modified in chan_usbradio.c was the product and vendor ID to match the 1.2 ones.So reversing this, if you can set these to CM108 in 1.3 you are already good to try it.I plan to try this today and will let you know the outcome Sent from my iPhone

Murf, 2024-03-02:
Simon rather than rebuild v1.3 rc1 with the CM108 Vendor and product ID , here is the changes to the ASL asterisk module :
https://github.com/Murf/ASL-Asterisk/commit/91b621e45c8ade2592f2a5e41bfa3558272de0bd

Murf, 2024-03-02:
Ok I see no difference with V1.3 :(

Murf, 2024-03-02:
UPDATE: and facepalm... I was using a Quansheng UV-K6, and have now swapped to a 'feng UV-5R and am progressing.

Murf, 2024-03-02:
And now OPERATIONAL :)
TL;DR
Use a Baofeng UV-K5, make sure you either 1) Set the Vendor/Product ID of the AIOC to 0d8c:000x OR Modify AllStarLink asterisk module chan_usbradio.c to use the normal AIOC IDs of 1209:7388.

skuep, 2024-03-03:
> UPDATE: and facepalm... I was using a Quansheng UV-K6, and have now swapped to a 'feng UV-5R and am progressing.

What exactly didn't work? I think those should be very similar 🤔

> And now OPERATIONAL :)

Nice! With virtual COS? Or are you using another way of detecting incoming voice?

> Modify AllStarLink asterisk module chan_usbradio.c to use the normal AIOC IDs of 1209:7388.

Just curious, how did you do it? Using the python script? Note that the next time you use the script, you need to adjust the VID/PID to the new one, or it doesn't find the AIOC.

Murf, 2024-03-03:
For the Quansheng, when I run the radio-tune-menu option "F" which is supposed to turn on PTT, emit a tone, then turn off PTT, which it does for the Baofeng UV-5R, but for the Quansheng UV-K6 the PTT stays on and never goes off, and no tone is emitted.

For the Vendor/Product ID detection I didnt use the script as I could not get it to run. Modifying the ASL chan_usbradio module itself to look for AIOC IDs rather than CM108 IDs is what I did.
So no serial port involved at all, just pure CM108 emulation.

For COS I didnt really do anything to the AllStarLink config other than what the Youtube I linked suggested, which is basically just set carrierfrom=VOX in usbradio.conf

May I ask, what exactly did you base the CM108 emulation on? Was it the PDF that ASL use for modifying the CM08 hardware?

Shopstream204, 2024-03-10:
Has anyone been able to successfully use the script?  I have tried on both a linux and windows machine and each time it fails.

Murf, 2024-03-10:
> Has anyone been able to successfully use the script? I have tried on both a linux and windows machine and each time it fails.

Which script?

Shopstream204, 2024-03-10:
The script for v1.3 RC1 that allows the vid and pid to be changed

Murf, 2024-03-10:
> The script for v1.3 RC1 that allows the vid and pid to be changed

No I could not get it to run either, errors in the file.
I just recompiled chan_usbradio.c with correct IDs

Shopstream204, 2024-03-10:
What all is involved in recompiling the file?  Once the file is compiled is it as easy as placing the file on your existing Allstarlink install or do you need to do a complete reinstall of Allstarlink?

Murf, 2024-03-10:
> What all is involved in recompiling the file? Once the file is compiled is it as easy as placing the file on your existing Allstarlink install or do you need to do a complete reinstall of Allstarlink?

You will need to install build tools on the computer that you have ASL installed on, what OS is it? and distribution?
Then you will need to download the version of ASL you are using (for example I am using ASL V2 beta)
Once you can confirm you can build it (not install it) then edit the chan_usbradio.c file to make the changes, as shown above in previous posts.
then just copy the module "chan_usbradio.so" to /usr/lib/asterisk/modules/, and restart asterisk or reboot

Shopstream204, 2024-03-18:
@Murf  I am using the ASL Beta 2 ISO.  I believe it's based on Debian 10.

Murf, 2024-03-18:
> @Murf I am using the ASL Beta 2 ISO. I believe it's based on Debian 10.

Perfect, thats what I am using.

Hadgee, 2024-03-24:
Hi, @Murf Brett, 

Is the modified 'chan_usbradio.c' (chan_usbradio.so) file available?. I have been struggling and only running Raspberry Pi on Linux and would like to try it. I have no major knowledge but can get the file to the correct directory. I would love to get my AIOC working on AllStar. Thanks in advance, or more information (laymans terms) on how to do it.
Ellis

Murf, 2024-03-25:
The file is available in my github, but you need to be able to build it. If
you are not familiar with this process then it is probably not the best
option for you.

Hadgee, 2024-03-25:
@Murf  No problem. I will investigate further.

Hadgee, 2024-03-30:
@Murf 
In the interest of my sanity....after losing far too many hours with permission errors and getting nowhere would it be possible to have a copy of your 'chan_usbradio.so' file as I believe it is only the equivalent of a library file.? I will try anything before assigning my AIOC cable to chirp and APRS and moving on from AllStarLink home repeater.
Yours Hopingly  Ellis de M5AEI

Murf, 2024-03-30:
> @Murf In the interest of my sanity....after losing far too many hours with permission errors and getting nowhere would it be possible to have a copy of your 'chan_usbradio.so' file as I believe it [...]

Assuming you are using a Pi, here you go
[chan_usbradio.zip](https://github.com/skuep/AIOC/files/14813585/chan_usbradio.zip)
Also here:
https://archive.org/download/chan_usbradio/chan_usbradio.zip

Hadgee, 2024-03-31:
@Murf  Thank you, yes Pi. I will let you know how it goes.

Hadgee, 2024-04-01:
@Murf 
Thanks again. All good first contact through to Hubnet today. AIOC V1.3.   Pi4 8gb running ASL 2.0Beta. I am using Baofeng UV-82. 
I have a bit of tweaking but works a treat. :-)

Murf, 2024-04-01:
Glad to help!

G1LRO, 2024-04-03:
Hi @Murf do you know if the SimpleUSB option been accomplished using the change in USB IDs and @skuep's virtual-COS? So just using a reconfiguration of the AIOC? It would be good to achieve this as SimpleUSB is much easier for novice users to do on AllStarLink.
It's not clear from the above thread, but it looks like USBRadio is the only config that's been made to work. 
Cheers Mark

Murf, 2024-04-03:
> Hi @Murf do you know if the SimpleUSB option been accomplished using the change in USB IDs and @skuep's virtual-COS? So just using a reconfiguration of the AIOC? It would be good to achieve this as [...]

I think it is best to wait for the ability to change the ID's on the AIOC firmware itself, then no code changes will be needed for asterisk.
@skuep do you have an ETA on a working python script to set AIOC USB Vendor/Device IDs please?

G1LRO, 2024-04-03:
Hi @Murf I think the USB ID manipulation was delivered and closed on https://github.com/skuep/AIOC/issues/46 

So hopefully we are nearly there. I'm not skilled in the manipulation of the AIOC registers so it would be nice to have a canned script to perform the AIOC register mods to make it SimpleUSB compatible. (setting USB ID and Virtual COS)

Murf, 2024-04-03:
Yes, that was the plan via  python script

Shopstream204, 2024-04-04:
There was a script provided  for RC1.3 but I have not had success using it.  That said I have no python knowledge and I am relatively green with Linux.

Murf, 2024-04-04:
Yes I believe that script was incomplete.
I could be wrong, but I expect its functionality will be something like:
./scriptname.py V_ID=0x0123 H_ID=0xABCD
pretty straightforward...

G1LRO, 2024-04-05:
Hi  @skuep , please do you have a moment to draft a script to configure the AIOC for AllStar as described above?

Hadgee, 2024-04-06:
@G1LRO Thank you for your firmware update script. I flashed my AIOC to v1.3 using it.

skuep, 2024-04-06:
You are correct. It's a pretty advanced feature with some caveats, that's why there is no canned script available yet.

````
import sys
import hid
from struct import Struct
from enum import IntEnum, IntFlag

class Register(IntEnum):
    MAGIC = 0x00
    USBID = 0x08
    AIOC_IOMUX0 = 0x24
    AIOC_IOMUX1 = 0x25
    CM108_IOMUX0 = 0x44
    CM108_IOMUX1 = 0x45
    CM108_IOMUX2 = 0x46
    CM108_IOMUX3 = 0x47
    SERIAL_CTRL = 0x60
    SERIAL_IOMUX0 = 0x64
    SERIAL_IOMUX1 = 0x65
    SERIAL_IOMUX2 = 0x66
    SERIAL_IOMUX3 = 0x67
    VPTT_LVLCTRL = 0x82
    VPTT_TIMCTRL = 0x84
    VCOS_LVLCTRL = 0x92
    VCOS_TIMCTRL = 0x94

    
class Command(IntFlag):
    DEFAULTS = 0x10
    RECALL = 0x40
    STORE = 0x80

class PTTSource(IntFlag):
    NONE = 0x00000000
    CM108GPIO1 = 0x00000001
    CM108GPIO2 = 0x00000002
    CM108GPIO3 = 0x00000004
    CM108GPIO4 = 0x00000008
    SERIALDTR = 0x00000100
    SERIALRTS = 0x00000200
    SERIALDTRNRTS = 0x00000400
    SERIALNDTRRTS = 0x00000800
    VPTT = 0x00001000

def read(device, address):
    data = device.get_feature_report(int(address), 5)
    address, value = Struct('<BL').unpack(data)
    return value

def write(device, address, value):
    data = Struct('<BL').pack(address, value)
    device.send_feature_report(data)

def dump(device):
    for r in Register:
        print(f'Reg. {r.value:02x}: {read(device, r.value):08x}')

# AIOC is searched for on the system using these values
aioc = hid.Device(vid=0x1209, pid=0x7388)

magic = Struct("<L").pack(read(aioc, Register.MAGIC))

if (magic != b'AIOC'):
    sys.exit(-1)

print(f'Manufacturer: {aioc.manufacturer}')
print(f'Product: {aioc.product}')
print(f'Serial No: {aioc.serial}')
print(f'Magic: {magic}')

if True:
    # Load the hardware defaults
    print(f'Loading Defaults...')
    write(aioc, 0, Command.DEFAULTS)

if True:
    # Set USB VID and PID (use with caution. Will need changes above to be able to re-configure the AIOC)
    vid = 0x1209 # Set VID here
    pid = 0x7388 # Set PID here
    write(aioc, Register.USBID, (vid << 0) | (pid << 16))
    print(f'Now USBID: {read(aioc, Register.USBID):08x}')

if True:
    # Store settings into flash
    print(f'Storing...')
    write(aioc, 0, Command.STORE)
```

This should do the trick. You can set the VID/PID to whatever values you like in the commented lines. After the script has stored the settings on the AIOC, you need to disconnect and reconnect it to your computer.

One important thing: If, for some reason you want to run the script again (e.g. reverting back the AIOC to defaults), you need to change the VID/PID to the new values in the commented line at the top. Otherwise the script will not find the AIOC (because it expects it to be at 0x1209/0x7388). In addition, you should remove all other CM108 devices, that could have the same VID/PID in order for the script not to confuse it with an AIOC.

Second thing: In another issue, we found some problems in conjunction with running the script on Windows. This has to do with Windows being a bit more accurate with what commands it can send to the AIOC (or any HID device). I hope I can find a nice solution for this in the future. For now, you need to use Linux or a Raspberry Pi.

G1LRO, 2024-04-06:
Thanks for this @skuep it's really helpful. Can we confirm that the script also enables the Virtual-COS function at the same time?

skuep, 2024-04-06:
Sorry, I forgot to include it, I will update it later

skuep, 2024-04-06:
I was wrong, I think virtual COS could be enabled by default on the CM108 volume up or down button.

G1LRO, 2024-04-06:
Thanks @skuep  it does appear the virtual COS is on by default following my initial testing. So far so good.

To help others trying to configure, I have set these in the SimpleUSB tune menu.:

COSFROM Mode (currently "usbinvert")
PTT Mode (currently active LOW)

I found the radio went into PTT on boot up of the AllStarLink so had to change the PTT mode. The COS mode in SimpleUSB is also inverted by default.

I think there are other settings to change to optimize performance, this is just a get-you-going.

Murf, 2024-04-07:
I have finally managed to get the python script to run (hint for when using the ASL2 pi image, command is python3 scriptfilename.py, and I needed pip3 install hid--1.0.4)

Unfortunately it does not work:

root@asl2-46471:/home/repeater# python3 aioc_params.py
Traceback (most recent call last):
  File "aioc_params.py", line 42, in <module>
    aioc = hid.Device(vid=0x1209, pid=0x7388)
  File "/usr/local/lib/python3.7/dist-packages/hid/__init__.py", line 130, in __init__
    raise HIDException('unable to open device')
hid.HIDException: unable to open device

root@asl2-46471:/home/repeater# lsusb
Bus 001 Device 005: ID 1209:7388 Generic

Murf, 2024-04-07:
UPDATE: I got the python script to work, I had to:
yum install hidapi
as well as do it on another linux box

skuep, 2024-04-07:
That's odd.. it should work on the raspberry pi. Did they do some fundamental changes to the ASL image? Is "root" really the root user (und 0) or is it a regular user just makes root? You could try using sudo with the python3 command, just to be sure...

Murf, 2024-04-07:
Its strange, after I change the IDs to CM108 it now lets me use the script
(once I updated the ID's to open of course)

skuep, 2024-04-07:
So you changed the USB IDs on your other Linux system and now it works on the ASL image? Hmmm.. still smells a bit like permission problems.

Murf, 2024-04-07:
Agreed :)

G1LRO, 2024-04-09:
I'm having trouble with the virtual-COS on ASL, I had it working the other day and now it's not working and the simpleusb tune menu shows no activity either. I can't see anything has changed.
Does anyone know how to verify the operation of the Virtual-COS buy viewing the expected signals that would come from the CM108 emulation?

Murf, 2024-04-09:
I am also having trouble with simpleusb signal detection. With usbradio I use the Vox setting but that is not an option with simpleusb

G1LRO, 2024-04-09:
I am have resolved the COS issue, it was  hardware problem elsewhere.

I am able to use SimpleUSB rather than Usbradio. To do this I have amended the USB ID info to the CM108 settings, it then works on simple usb. set carrierfrom = usbinvert  and ctcssfrom = no . Also set invertptt = 0 

This uses the Virtual-COS Simon has written into the AIOC to avoid using USBradio and Vox, I tried USBradio before with David's modified library for the AIOC but couldn't make it work,
Hope this helps

G1LRO, 2024-04-10:
I'm struggling with the python script on a fresh raspbian 11 (Bullseye) install. getting the same 'unable to open device' error:
```
`Traceback (most recent call last):
  File "/home/admin/aioc.py", line 42, in <module>
    aioc = hid.Device(vid=0x1209, pid=0x7388)
  File "/usr/local/lib/python3.9/dist-packages/hid/__init__.py", line 130, in __init__
    raise HIDException('unable to open device')
hid.HIDException: unable to open device
`
```
Now i think this has something to do with permissions as I tried the commands in this article https://bbs.archlinux.org/viewtopic.php?id=278341

It made the python script work perfectly for several times. Until, when I put the commands into a script and ran it again, it's all broken and I can d work out how to get it going again. 

I have deleted the /etc/udev/rules.d/92-viia.rules created by these commands and tried running from scratch, no joy.

Any Ideals folks?

Murf, 2024-04-11:
I had this issue and had to change the ID from another linux host (Intel
nuc). Pi would not work.

Murf, 2024-04-11:
I have used the simplusb settings following, and everything works except DTMF is not detected.
carrierfrom = usbinvert
ctcssfrom = no 
invertptt = 0

G1LRO, 2024-04-11:
> I had this issue and had to change the ID from another linux host (Intel nuc). Pi would not work.
> […](#)
>

Interestingly I have one Pi set up (v11 Bullseye) where this works consistently no problem, but all the new images I am trying won't work so there is a solution somewhere. @skuep any ideas?

G1LRO, 2024-04-11:
> I have used the simplusb settings following, and everything works except DTMF is not detected. carrierfrom = usbinvert ctcssfrom = no invertptt = 0

DTMF was working for me, but the virtual-cos was dropping between numbers  so you have t send quickly. I tested with the Asterix CLI as in this vid: https://www.youtube.com/watch?v=iCZsMFrVJlM&t=1485s

skuep, 2024-04-11:
> > I had this issue and had to change the ID from another linux host (Intel nuc). Pi would not work.
> > […](#)
> 
> Interestingly I have one Pi set up (v11 Bullseye) where this works consistently no problem, but all the new images I am trying won't work so there is a solution somewhere. @skuep any ideas?

Can you try running the script with ``sudo``?

skuep, 2024-04-11:
> DTMF was working for me, but the virtual-cos was dropping between numbers so you have t send quickly. I tested with the Asterix CLI as in this vid: https://www.youtube.com/watch?v=iCZsMFrVJlM&t=1485s

Nice! I think the thresholding detector on the AIOC needs some tuning still. There are ways to manipulate the timeouts using the script as well, but I have not had time to define the registers for that yet.

skuep, 2024-04-11:
> I have used the simplusb settings following, and everything works except DTMF is not detected. carrierfrom = usbinvert ctcssfrom = no invertptt = 0

Is invertptt = 0 special (different from default)? From what I can tell from e.g. https://allstarsetup.com/modify-a-cm108-sound-fob/ invertptt should have the same polarity as the AIOC.

What about carrierfrom=usbinvert? The AIOC sets the Volume UP or DOWN button (not sure) when the carrier is received. I think that is the other way around by default in ASL?

[23 more replies]