skuep/AIOC #62: Retevis RT73

dc7os, 2024-03-09:
Could i program the RT73 with a computer running Win 10/11?

73 de Karsten DC7OS

skuep, 2024-03-24:
That's a good question! I am not quite familiar with this radio and it's hard for me to found out just from the search results. I think your best bet is to join the discord channel and ask there. There are a lot of experienced people in terms of HTs. Win10/11 shouldn't be a problem.

dc7os, 2024-03-24:
> That's a good question! I am not quite familiar with this radio and it's hard for me to found out just from the search results. I think your best bet is to join the discord channel and ask there. There are a lot of experienced people in terms of HTs (Is it even a HT?). Win10/11 shouldn't be a problem.
> [Verfolgungsbild][https://github.com/notifications/beacon/ACQAJYUJFLU2EEXEZGMZOZTYZ244VA5CNFSM6AAAAABEOCSOXKWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTYGWWLA.gif]

Ich jann Dir sagen, es tut einwandfrei mit Win11 und dem RT73

closed 2024-04-06 by skuep