skuep/AIOC #63: AprsDroid PTT request

oe9fwv, 2024-03-19:
I tried my Anytone 878 with Aprsdroid. Receive and transmit is working when  I use an OTG cable and press the PTT when I want to transmit a beacon. (Caveat: RF power should be kept low to avoid PTT locking) The problem is that VOX does not work when the input is from the mic connector and not from the built in microphone. The only comfortable way to transmit would be a DATA vox function in the AIOC, if Aprsdroid does not provide a PTT function in the future.

skuep, 2024-03-24:
Yeah, that is a known issue. There is currently little work in aprsdroid to implement a PTT mechanism (such as COM or CM108). See this issue here: https://github.com/ge0rg/aprsdroid/issues/324

There is a pre-release AIOC firmware available, that implements Virtual PTT (a version of VOX). See this issue for more information: https://github.com/skuep/AIOC/issues/19

penguin359, 2024-07-26:
I have my CM108 now and have confirmed that the PTT control via RTS is working from Android and I'm hoping to help get this added as part of ge0rg/aprsdroid#324. The current implementation from that ticket does not yet seem to support this specific chip, but other libraries do seem to work with it.

skuep, 2025-01-04:
I think there was some recent work on aprsdroid to get this supported. So far I am only reading about FTDI and digirig. Have you tried the new aprsdroid version using the AIOC by chance?

RyanHope, 2025-01-07:
I have used the config script to set vid=0x0d8c and pid=0x000c as well as enabled the AutoPTT. I have tested this with a Quansheng K6 UVK5, Baofeng BF-F8HP PRO and a TIDRADIO TH-H3... With VOX off, the PTT is properly getting triggered. The volume on my phone cannot be over 50% or else clipping will occur and I will get no audio transmitted; all I hear is chunk on the monitoring radio. Under 50% the signal is clear. However, I have not got APRSdroid to decode any data received, not even my own beacon, not with any radio. So, got half the puzzle figured out.

skuep, 2025-01-07:
Could be an intermittent connection on the TRS connector solder joint. Can you check for conductivity between the TRS contacts and the PCB solder joints?

oe9fwv, 2025-01-07:
> I think there was some recent work on aprsdroid to get this supported. So far I am only reading about FTDI and digirig. Have you tried the new aprsdroid version using the AIOC by chance?

I use APRSdroid 1.6.3d 2024-04-03 Version.

This seems to be the latest version. I use it with the VOX Firmware of AIOC. 
It works, settings in Aprsdroid are: 
Audio (AFSK)
high quality decoder checked
Audio out set to Audio-Call
Sync Prefix 500 msec

73! Werner oe9fwv

oe9fwv, 2025-01-07:
> @oe9fwv thanks for the information! In this issue particularly we are looking for having "real" PTT with the AIOC instead of the virtual PTT (aka VOX) bandaid

I do not see a possibility for a real PTT in Aprsdroid. Should I?

skuep, 2025-01-07:
Why wouldn't it? See the link to the corresponding aprsdroid issue in the second post. There is a beta version that allows using RTS (at least with FTDI adapters and digirig)

oe9fwv, 2025-01-07:
> Why wouldn't it? See the link to the corresponding aprsdroid issue in the second post. There is a beta version that allows using RTS (at least with FTDI adapters and digirig)

thank you for the link to the new version of Aprsdroid.
I did not see the new version in the Appstore. I tried it now. With the VOX Version of AIOC it works like before, with the normal AIOC it does not transmit. Receive works very well (with the Anytone AT898) The Anytone seems to be better in receiving than the Quansheng. 

If I missed some setting in Aprsdroid please let me know. btw, I do not have a map in aprsdroid with the new version :-(
73! Werner