skuep/AIOC #64: Suggestions on what to use to short pins for flashing?

ITCMD, 2024-03-28:
The pins next to the USB-C connecter that must be shorted for flashing are smaller than anything I have readily available. Ive tried staples, GPIO male pins, enamel wire, etc, nothing has been small enough without grinding them down.

What I have been using is a bent spring I have, but it results in finger prickings often - redefining blood and sweat into your work!

skuep, 2024-03-29:
They are designed for 1.27mm pin headers originally. Paperclips seem to work as well, or sharp tweezers.

Anyway, you really only need this once for initial programming (and technically not even there, because without a valid application, the STM32 goes to the bootloader immediately). With Firmware version 1.2 the AIOC is recognized by dfu-util and reset into bootloader by software.

ITCMD, 2024-03-30:
Gotchya. Im mostly doing all them for the first time before shipping them out. Ill try that with the newer firmware!

skuep, 2024-04-06:
Ah, I see. Just a hint: There are apparently some issues with the DFU Runtime not being loaded the correct driver under Windows. Some people reported that installing some kind of LibUSB Driver with the Zadig tool helps.

G1LRO, 2024-04-10:
Opening the prog and 3.3v holes to .75mm on the Gerber allows for the use of a breadboard wire with the pins on each end. These are nicely insulated and relatively safe.

ITCMD, 2024-04-20:
> Opening the prog and 3.3v holes to .75mm on the Gerber allows for the use of a breadboard wire with the pins on each end. These are nicely insulated and relatively safe.

Would you be able to upload this gerber?

skuep, 2024-04-21:
The holes are designed for regular 1.27mm pitch header pins, like this:
https://www.wayconn.com/products/board-to-board/1-27mm-pitch-pin-header/

ErikS-web, 2024-05-26:
I used a thin wire from a resistor. Works fine for my 5 devices.

After that `updating firmware can be done Without a Wire`. See [Firware updating](https://github.com/skuep/AIOC#firmware-updating)

W1BTR, 2024-07-20:
> I used a thin wire from a resistor. Works fine for my 5 devices.
> 
> After that `updating firmware can be done Without a Wire`. See [Firmware updating](https://github.com/skuep/AIOC#firmware-updating)

Im typically only installing on them for the first time. I ended up finding a jumper wire at work that works. It needs to be a perfectly round ones, not one of the cheap ones that have a + shape

closed 2024-07-20 by ITCMD