skuep/AIOC #65: Alternative USB-C connector

ITCMD, 2024-04-20:
The Amphenol ICC 12401832E402A has been out of stock for a bit, and ordering it through the global parts manager always results in a refund with no explanation.

What alternatives could I use? TE Connectivity 2305018-2? JAE Electronics DX07S024JJ3R1300?

skuep, 2024-04-21:
Hmm, those alternatives don't look like they have the same footprint. I am currently swamped with work, so I don't have the time to find a replacement. Maybe someone else can come up with one? You could try the discord group maybe.

lexszero, 2024-04-22:
Dealing with the same problem, I have changed USBC connector to U262-161N-4BVC11 (mostly because it was both in stock and in KiCad library). The order is in manufacturing now, so I don't know yet if the footprint is right (but it passed JLCPCB's DFM analysis fine).  Should I create a fork with my changes?

ITCMD, 2024-04-23:
> Hmm, those alternatives don't look like they have the same footprint. I am currently swamped with work, so I don't have the time to find a replacement. Maybe someone else can come up with one? You could try the discord group maybe.

Didnt even know there ws a discord group! Will take a look

> Dealing with the same problem, I have changed USBC connector to U262-161N-4BVC11 (mostly because it was both in stock and in KiCad library) and rerouted a few traces around it to fit. The order is in manufacturing now, so I don't know yet if the footprint is right (but it passed JLCPCB's DFM analysis fine). Should I create a fork with my changes?

That'd be much appreciated, I'll probably wait to hear on how your order went!

skuep, 2024-04-27:
I was hoping there is a USB-C connector with the same footprint... Hmm

lexszero, 2024-04-27:
> I was hoping there is a USB-C connector with the same footprint... Hmm

USB connectors are quite diverse in their footprints, and the one used in your design is also not the most common (two rows of pads for each side of USB-C, most popular have just a single row and it's usually enough if you don't need any fancy USB-C shenanigans). Position of mounting features with relation to contact pads also vary wildly, I tried to find a direct replacement part for an hour and gave up.

> I'll probably wait to hear on how your order went!

It went well, received it yesterday, it works! Shorting of BOOT to 3v3 was required to get it to flash, I'm now struggling with unrelated issue (2.5mm jacks are not exactly common, and those which I've managed to purchase locally are junk and ring terminal loses continuity inside the jack as soon as I get it soldered in place).

legonigel, 2024-04-29:
FYI, (as of less than 24 hours ago) JLC has these connectors back in stock.

ErikS-web, 2024-05-26:
Yes. this item can be closed.

I ordered suc6fulle 14 days ago.

Only problems was the `green led`. But could buy them from other JLCPCB-members.
Minimum order for that was $2, so I bought 57 green LED's  ;-)

skuep, 2024-06-02:
Nice! Yeah, it happens from time to time that something goes out of stock. So far there was nothing major missing (dodged a bullet with the USB connector there)..

I hope it's okay to close this issue in the meantime..

closed 2024-06-02 by skuep