skuep/AIOC #66: Problems getting HID libraries functioning in Python 3 on Raspberry Pi OS

G1LRO, 2024-04-22:
V1.3 firmware has the facility to manipulate registers in the AIOC CPU. The manipulation requires the use of HID libraries, in this case using a Python script.

Raspberry Pi OS produces errors when attempting run the python script.  For example:
```
admin@raspberrypi:~ $ sudo python3 aioc.py
Traceback (most recent call last):
  File "/home/admin/aioc.py", line 43, in <module>
    aioc = hid.Device(vid=0x1209, pid=0x7388)
AttributeError: module 'hid' has no attribute 'Device'
```
Both the 'hid' and 'hidapi' libraries are loaded (tried singularly and in combination).

**Replicating the issue:**
Load Raspbian (Bookworm or Bullseye) to a fresh SD card, boot into a Raspberry Pi (V3b or later)
Create the pyhton script 'aioc.py' from issue  https://github.com/skuep/AIOC/issues/61#issuecomment-2041084242 
load HID libraries - 
```
sudo apt install python3-hid
sudo apt install python3-hidapi

```
Run the script

`sudo python3 aioc.py`

I've tried many combinations of 32 and 64 bit OS and on Pi 3 and 4 hardware.

G1LRO, 2024-04-22:
The solution is to install just these libraries for the hid function:

```
sudo pip install hid
sudo apt install libfuse2 libhidapi-hidraw0 libhidapi-libusb0
```

Then run the script
Tested on 32 bit bullseye

closed 2024-04-22 by G1LRO

alexjj, 2024-06-13:
I've been following the instructions but the python script, aioc.py, gives me errors. I flashed the firmware fine with `flash1.3.sh`, and the light is on and it appear fine. 

Running `sudo python3 aioc.py` first gives the error:

```python
Traceback (most recent call last):
  File "/home/alex/aioc/aioc.py", line 43, in <module>
    aioc = hid.Device(vid=0x1209, pid=0x7388)
           ^^^^^^^^^^
AttributeError: module 'hid' has no attribute 'Device'. Did you mean: 'device'?
```

So I changed lined 43 to `aioc = hid.device(vid=0x1209, pid=0x7388)`

However, I then get the following error:

```python
Traceback (most recent call last):
  File "/home/alex/aioc/aioc.py", line 45, in <module>
    magic = Struct("<L").pack(read(aioc, Register.MAGIC))
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/alex/aioc/aioc.py", line 34, in read
    data = device.get_feature_report(int(address), 5)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "hid.pyx", line 187, in hid.device.get_feature_report
ValueError: not open
```

My device shows up as `ID 1209:7388 Generic All-In-One-Cable` with `lsusb`.

All these packages are installed:
```
sudo pip install hid
sudo apt install libfuse2 libhidapi-hidraw0 libhidapi-libusb0
```

I am running on a laptop with Debian 12 to do the flashing...

spommere, 2024-09-27:
Hi,

just received an AIOC frmo N6AD.

With all packages installed, and lsusb showing

$ lsusb
Bus 001 Device 004: ID 1209:7388 Generic All-In-One-Cable

I also get Broken pipe:

$ sudo python3 aioc.py
Traceback (most recent call last):
  File "/home/pi/g1lro/aioc.py", line 45, in <module>
    magic = Struct("<L").pack(read(aioc, Register.MAGIC))
  File "/home/pi/g1lro/aioc.py", line 34, in read
    data = device.get_feature_report(int(address), 5)
  File "/usr/local/lib/python3.9/dist-packages/hid/__init__.py", line 232, in get_feature_report
    size = self.__hidcall(
  File "/usr/local/lib/python3.9/dist-packages/hid/__init__.py", line 190, in __hidcall
    raise HIDException(err)
hid.HIDException: ioctl (GFEATURE): Broken pipe

Any suggestions ?

spommere, 2024-09-28:
Well, I flashed it with the v1.3.0 firmware,  now it's working:

Manufacturer: AIOC
Product: All-In-One-Cable
Serial No: b53bed37
Magic: b'AIOC'
Loading Defaults...
Now USBID: 000c0d8c
Now VCOS_TIMCTRL: 00005dc0
Storing...

skuep, 2024-09-28:
The USB interface has changed recently, due to issues with windows on the old one. You need to use the correct python script for the according firmware version. 

Where is the script from,  that you are running?