skuep/AIOC #70: Error while debugging on a nucleo f302

PhipsmonET, 2024-05-14:
Hey, i'm trying to implement this on a nucleo board but I'm facing some issue while launching the debug in CubeIDE. 
Basically after a succesful build I get this error in debug:

Error in final launch sequence:

Failed to execute MI command:
load C:\\Users\\pit\\Desktop\\stm32\\aioc-fw\\Debug\\aioc-fw.elf 

Error message from debugger back end:
Load failed
Failed to execute MI command:
load C:\\Users\\pit\\Desktop\\stm32\\aioc-fw\\Debug\\aioc-fw.elf 

Error message from debugger back end:
Load failed
Failed to execute MI command:
load C:\\Users\\pit\\Desktop\\stm32\\aioc-fw\\Debug\\aioc-fw.elf 

Error message from debugger back end:
Load failed
Load failed

Is this an already known issue? 
Thanks!

PhipsmonET, 2024-05-18:
I've just noticed that the bin size of my release bild is way bigger than the build that Skuep has given us, i'm sure im missing somenthing. 

P.S my bin file is 125Kb, way bigger than the flash size of the f302

skuep, 2024-05-18:
I am not exactly sure.. my files are a Release build... What linker script do you use?

PhipsmonET, 2024-05-19:
Ah thanks!, that might be the problem! I was indeed using the 302cb linker so no wonder it didn'work. 
I'll try to build with the 302r8 script and get back to you as soon as possible.

skuep, 2024-09-18:
I am going to close this issue, feel free to reopen if you have problems.

closed 2024-09-18 by skuep