skuep/AIOC #75: AIOC on Alinco DJ-VX50

AcidMal, 2024-06-21:
I'm trying to use the AIOC on my Alinco transceiver but it looks like the pinout on the radio is a tad bit different. I was wondering if this is something you guys might be able to help me with as I have tried to search for guides but am unable to see how I can fix this.

skuep, 2024-06-23:
Hi @AcidMal,
"The AIOC way" would be to create a different PCB layout that suits the connectors on the Alinco transceiver, but it is a bit involved. Do you have information on the connectors of the Alinco transceiver? Mechanical and Electrical maybe?

skuep, 2024-09-18:
Due to inactivity, I am going to close the issue. Feel free to reopen if required.

closed 2024-09-18 by skuep

ffeldner, 2025-07-19:
from what I could gleam, the ICOM ID-52/51A etc. Speaker Mics are compatible to the DJ-VX50, whereas the standard K-plug ones are permanently PTT on when plugged in.  ( reference: https://www.youtube.com/watch?v=MRMrkJf2V_A ) furthermore, the ICOM pins seem to be slightly closer to each other than a Kenwood plug would be. there is also probably a resistor in there that is missing in the standard speaker plugs that is responsible for the Alinco being on perma-send with one of those plugged in.

I found this: https://forum.digirig.net/uploads/default/original/1X/07bccfd5a9a6b2c36fa9ce1a0f7c122776f93d75.jpeg

and this https://www.youtube.com/watch?v=OjhWREo_D7k from the same source

for data communication with the CI-V mentioned in the digirig schematic picture, the schematic with the diode and the resistor in this could be compatible to the Alinco: https://n0dim.com/Documents/Build-a-OPC-478U-Clone-200727.pdf