skuep/AIOC #76: AIOC to work with Radioshack HTX-10

ecologito, 2024-07-04:
Hi, 

I would like to connect the AIOC to my 10 meter radio (radioshack htx-10) and potentially use it for FT8. Here is the mapping on the radio that uses GX16-6 connector: 

Pin 1 Mic Audio
Pin 2 Ground on receive (this is one of the speaker terminals
Pin 3 PTT ground to transmit
Pin 4 channel select (pin 6=Up Pin 3 through 22K =Down)
Pin 5 Ground
Pin 6 +13.8V

I already have a "motorola style" female connector with pigtails so I can go from the original AIOC connectors to bare cables. Any help will be much appreciated.

ecologito, 2024-07-10:
I would like confirmation from someone who knows more than I do but this is what I have so far: 

Connect TRS 1 (Speaker & Tx data) to the GX16 6-pin connector:

Tip (TRS 1): Not used.
Ring (TRS 1): Not used.
Sleeve (TRS 1): Connect to Pin 2 (Ground).

Connect TRS 2 (Mic & PTT) to the GX16 6-pin connector:

Tip (TRS 2): Not used.
Ring (TRS 2): Connect to Pin 1 (Mic Audio).
Sleeve (TRS 2): Connect to Pin 3 (PTT).

Ensure Pins 4, 5, and 6 are not connected to avoid any issues with selector channel, ground, or 13.6V DC.

skuep, 2024-07-14:
Unfortunately I have no information for these radios, pin outs and also regarding the voltage levels.. maybe you can try the discord channel, there are a lot of guys with non-HT radios

skuep, 2024-09-18:
Due to inactivity, I am going to close the issue. Feel free to reopen if required.

closed 2024-09-18 by skuep