skuep/AIOC #77: PTT1 vs PTT2

W1BTR, 2024-07-09:
How does one control PTT1 vs PTT2, is it DTR vs RTS, or are they both triggered simultaneously?

skuep, 2024-07-14:
Hi there. Currently (by default) PTT2 is mapped to CM108 GPIO4. See here
https://github.com/skuep/AIOC/blob/5dbe16d50b62e1c1b01059a1a592ad8fa60a22d5/stm32/aioc-fw/Src/settings.h#L51

There is a way to change this configuration in the pre release firmware. There are some scripts in the issue tracker somewhere showing how to change the PTT mapping. Note that they need the pre release software and you need some Linux device for running the scripts (currently).

W1BTR, 2024-07-14:
> Hi there. Currently (by default) PTT2 is mapped to CM108 GPIO4. See here
>
> https://github.com/skuep/AIOC/blob/5dbe16d50b62e1c1b01059a1a592ad8fa60a22d5/stm32/aioc-fw/Src/settings.h#L53
>
> There is a way [...]

Im more curiois on how it is used kn the computer side. How does computer software support the dual ptt feature, and what does it change on the output?  Does it short the ptt2 connection to ground or do something else?

Or are you saying that it cannot currently be changed on the fly, but instead must be changed with the firmware or scripts?

skuep, 2024-07-14:
Yes, neither RTS nor DTR is currently "connected" to the PTT2 pin (the reason for that can be read in issue #11). Only the CM108 GPIO4 can be used for PTT2 (by default).

I am open in changing the default for PTT2 if it does not collide with findings in the issue #11. In summary, Windows is always setting those RS232 control pins willy-nilly (I think something about probing for serial mouses?) and it was hard to find a combination (for PTT1) that is "quiet" on all operating systems, when no actual ham software is in use.

skuep, 2024-09-18:
Due to inactivity, I am going to close the issue. Feel free to reopen if you need more help.

closed 2024-09-18 by skuep