skuep/AIOC #79: "Pop" sound on RX start and end

W1BTR, 2024-07-20:
When the squelch opens for RX, there is a loud "pop" sound that is a good 6db louder than any signal. This "pop" sound also occurs at the end when the squelch closes.

Anyone know of any good ways to isolate the audio circuit to reduce this noise? One possible use for the AIOC is to make a repeater, but this pop sound makes it difficult.

skuep, 2024-07-21:
You have to give me some clues about your setup... It might be that the radio switches off the DC level on the speaker output during idle if I understand you correctly. Then each time it switches the amplifier on it off, the AC decoupling cap needs to be charged again.

I am not sure if there is anything I can do there. 🤔

W1BTR, 2024-07-21:
Hm not sure what's causing it exactly. On all radios Ive tried (UV-5R, UV-K5, GA-510, and some custom connected radios like the ic-2820h) this happens with the AIOC going into voicemeeter or a recording software, but not when the radio is connected to a speaker or speaker mic. Ill have to record a vid at some point

skuep, 2024-09-18:
If you can, it would be best to post a picture or video of your setup and the problem. In the meantime I am closing this issue due to inactivity, but you can reopen it anytime in the future.

closed 2024-09-18 by skuep