skuep/AIOC #80: STM2f303CCTx

Timmotools, 2024-07-20:
Hi,

Love your work here and I have a board coming soon. 

I have a question though. Would the firmware run on the STM32F303CCTx chip? 

While I am waiting, I was going through my junk boxes and found an old drone SPracing F3 flight controller. These have the STM32F303CCTx chip. 

I have mapped out all of the pins needed for the mappings and all the pins I need are going to solder pads. So little hardware modding and adding a few components. It should work.

Thanks.
[image]
[image]

Timmotools, 2024-07-21:
Just an update on this. I have flashed it with the latest firmware with stmcube via the built in usb to UART connection usually used for flashing or configuring cleanflight or betaflight. Then wired up the usb interface with a scrap usb lead. I can see in dev manager the sound and serial interfaces. 

This looks promising. 

It's just after midnight here in Australia and I have to work in the morning. Tomorrow I'll have a play around with the rest of the extra components and see if it works. 

Cheers
VK3FTP.

skuep, 2024-07-21:
Nice! I think as long as the chip has an ADC and DAC, you should be golden. Remember to select the correct part for the compilation process and you might need to swap the linker script for your MCU.

But if you're already seeing the USB devices, you should be almost there 😁 you will need some circuitry for the analog stuff though (level shifting, ac coupling and signal attenuation). Also for PTT it's best of you also use some simple bipolar NPN transistors.

Just a heads up, in the past I had a noisy signal at 1khz due to the USB interface on the MCU being on the same power supply, which is why I chose a dual regulator with one output for analog and one for digital power.

Timmotools, 2024-07-25:
Just an update on this. I have built all of the level converters and ptt control lines (single ptt) as per your schematic. The audio in/out and ptt are working fine. However the serial lines are not.

I did have to remove a 3.3v pull-up and short a series resistor on the AF-OUT (vbat) line.

Here is a photo of the board. Definitely not pretty but it's not a permanent solution. More a proof of concept.

[image]

I tried compiling the firmware with my chip selected but same issue. Chirp says short header.
On some more research on the board I found that the USART1 line is also the same usart line used for the built on usb to UART chip. So this is probably causing issues. 

I'll change the code and try later.

skuep, 2024-07-26:
Oh, yeah that could be a problem. You could change to another UART maybe in this case!

skuep, 2024-09-18:
Due to inactivity, I am going to close the issue. Feel free to reopen if required.

closed 2024-09-18 by skuep