skuep/AIOC #82: Link to download driver for windows

bshambaugh, 2024-07-31:
The AIOC driver for windows is hard to locate. The link appears to be dead. Could someone help me find it?

skuep, 2024-07-31:
There is no "AIOC driver". In all supported OSes (Linux, Mac, windows) the driver for the soundcard, serial interface and HID device are included and should be installed automatically. Apparently sometimes windows doesn't automatically install the drivers, but I have no clue why (I am not a windows user).

What I heard however is that it can be tricky to get the USB du Driver working on Windows systems. Some People reported to have opted to use a RasPi for dfu programming.

skuep, 2024-09-18:
Due to inactivity, I am going to close the issue. Feel free to reopen if required.

closed 2024-09-18 by skuep