skuep/AIOC #83: Ordering Amphenol ICC 12401832E402A USB C Connector through global parts - MouseReel, Tape & Reel, Cut Strips, or Cut Tape?

W1BTR, 2024-08-01:
No stock of the USB C connector again, so gonna have to get it through global parts sourcing.

I see a lot of options for how to get the parts. I know cut tape is what we do for the SM32 chip, but does it matter for the USB C connector? Other options such as MouseReel are cheaper for small quantities.

WROG208, 2024-08-02:
Just be careful what you order. I ordered 200 pcs and was quoted $180 then a week later the quote changed to an extra $200 and change additional to what i had already paid and no way to back out . 
Never had that problem before and i reported the seller as a bait and switch but nothing has been said. The irder date also changed from 3 weeks to 3 months delivery time.

W1BTR, 2024-08-02:
That sucks! Do you know what seller you used?

WROG208, 2024-08-02:
I will have to check. Edge key was the other seller u was going to buy from but their shipping was 4 months.

No ither private seller to buy from. Wish there was another usb c we could buy but i didn't find 1 with the same characteristics as the Amphenol

WROG208, 2024-08-05:
> That sucks! Do you know what seller you used?

Received the Amphenol USB today. Trued looking for the seller information but at least on the phone it will not show that option.
Have to get home and make an order.

W1BTR, 2024-08-06:
Did you get it shipped to you or to JCLPCB? I'll be ordering it through them and shipping it there.

WROG208, 2024-08-06:
JCLPCB took a while to deliver but they got there.

skuep, 2024-09-18:
Due to inactivity, I am going to close the issue. Feel free to reopen if required.

closed 2024-09-18 by skuep