skuep/AIOC #84: AIOC Audio Out Volume Playback Volume] ch = 1, val = -24576/0/1

lha, 2024-08-04:
I'm trying to use AIOC on **Ubuntu 24.04 LTS** and its once PTT happen, direwolf disconnects with audio error and PTT is left high (perhaps if the hist disconnect, PTT should be inactive ?)

```
[  307.293734] usb 1-1.2: Warning! Unlikely big volume range (=24576), cval->res is probably wrong.
[  307.293767] usb 1-1.2: [2] FU [AIOC Audio Out Volume Playback Volume] ch = 1, val = -24576/0/1
[  307.300986] usb 1-1.2: Warning! Unlikely big volume range (=24576), cval->res is probably wrong.
[  307.301012] usb 1-1.2: [2] FU [AIOC Audio Out Volume Playback Volume] ch = 1, val = -24576/0/1
[  307.313723] usb 1-1.2: Warning! Unlikely big volume range (=24576), cval->res is probably wrong.
[  307.313742] usb 1-1.2: [18] FU [AIOC Audio In Capture Volume] ch = 1, val = -24576/0/1
[  307.320479] usb 1-1.2: Warning! Unlikely big volume range (=24576), cval->res is probably wrong.
[  307.320497] usb 1-1.2: [18] FU [AIOC Audio In Capture Volume] ch = 1, val = -24576/0/1

```

**Direwolf** say

```
PTT 0 = 0
Audio input device 0 error code -19: No such device
Audio input device 0 error code -19: No such device
Audio input device 0 error code -19: No such device
```

skuep, 2024-08-05:
There are some known issues that have to do with EMI causing the USB system to restart. Can you try reducing output power or use a well matched antenna  (dipole or monopole with tiger tail) to narrow the issue down?

See https://github.com/skuep/AIOC/issues/11

skuep, 2024-09-18:
Due to inactivity, I am going to close the issue. Feel free to reopen if required.

closed 2024-09-18 by skuep