skuep/AIOC #86: AIOC and SVXLINK

F8ASB, 2024-09-11:
Hello all,

I'm trying AIOC with Svxlink repeater software, 
AIOC is detected and audio level on alsamixer are available and /dev/hidraw0 is present on my Raspberry Pi.

**For RX config** is for a CM108 and SQUELCH(COS) detection on VOL_DW pin 48.
HID_DEVICE=/dev/hidraw0
HID_SQL_PIN=VOL_DN
SQL_DET=HIDRAW

**For TX config**:
PTT_TYPE=Hidraw
HID_DEVICE=/dev/hidraw0
HID_PTT_PIN=GPIO3

**I have this error:**
Wed Sep 11 10:04:09 2024: *** ERROR: unknown/unsupported sound chip detected...
Wed Sep 11 10:04:09 2024: *** ERROR: Squelch detector initialization failed for RX "Rx1"
Wed Sep 11 10:04:09 2024: *** ERROR: Could not initialize RX "Rx1"
Wed Sep 11 10:04:09 2024: *** ERROR: Could not initialize Logic object "SimplexLogic". Skipping...

_Have you an idea to resolve this problem?_ 

Maybe HID_SQL_PIN=VOL_DN isn't correct for AIOC.

F8ASB, 2024-09-18:
Hi!,

Thanks for your reply.

I'll try and report back on this thread.

tomaz1, 2024-12-16:
Hi,

I have AIOC on SvxLink, and I can confirm the solution from [G1LRO](https://g1lro.uk/?p=676) works.
I have also tried [aioc-fw-1.3.0-rc.2.bin](https://github.com/skuep/AIOC/releases/download/v1.3.0-rc.2/aioc-fw-1.3.0-rc.2.bin) and setting the VID and PID to "Bus 001 Device 006: ID 0d8c:000c C-Media Electronics, Inc. Audio Adapter" with the Python script provided in the release notes: write(aioc, Register.USBID, (0x0d8c << 0) | (0x000c << 16)) also works!

Thanks @F8ASB for the settings provided:

Rx settings:
HID_DEVICE=/dev/hidraw0
HID_SQL_PIN=VOL_DN
SQL_DET=HIDRAW

Tx settings:
PTT_TYPE=Hidraw
HID_DEVICE=/dev/hidraw0
HID_PTT_PIN=GPIO3

It works, but with both solutions, when the station stops transmitting, I get Rx triggered after approximately 500ms, and the signal (silence) lasts around 300ms. (Rx is sending a short silence to SvxLink, and the AIOC green light is brighter, indicating it is receiving something.)

Why is Rx getting triggered (immediately) after Tx?

[My current, not-so-good workaround/solution](https://github.com/tomaz1/PMR-FRN-SvxLink/blob/main/Navodila/AIOC-upgrade-FW-to-1.3.md) is to have two Rx settings and use "[Voter](https://www.svxlink.org/doc/man/man5/svxlink.conf.5.html)" where I can set VOTING_DELAY=600. 

P.S.: SvxLink runs in userspace as the user "svxlink", so it needed permissions to use /dev/hidraw0, like this:

```
cat /etc/udev/rules.d/99-hidraw.rules
KERNEL=="hidraw*", SUBSYSTEM=="hidraw", MODE="0660", GROUP="plugdev"
#Or if I would allow only (ID 0d8c:000c)
#KERNEL=="hidraw*", SUBSYSTEM=="hidraw", ATTRS{idVendor}=="0d8c", ATTRS{idProduct}=="000c", MODE="0660", GROUP="plugdev"

sudo usermod -aG plugdev svxlink

sudo udevadm control --reload-rules
sudo udevadm trigger

```

skuep, 2025-01-04:
I suspect you are using the VCOS for detecting incoming audio correct? From what I have heard about VCOS it tends to have issues with spurious activations. In current versions it will be possible to hardware a pin on the AIOC (e.g. unused PTT2) as input so you can tap a COS signal off the radio and connect it with the AIOC for this purpose.

For now I am closing this issue, but I thank you for sharing your experiences

closed 2025-01-04 by skuep

trygvis, 2026-11-14:
Instead of changing the AIOC firmware, I changed svxlink to use the serial port control pins instead:

```patch
-PTT_TYPE=Hidraw
-# Symlink created by run.sh
-HID_DEVICE=run/aioc-hid
-HID_PTT_PIN=GPIO3
+# PTT via the AIOC's virtual serial port: it keys on DTR=1 and RTS=0.
+# (Hidraw PTT doesn't work: svxlink only accepts C-Media USB IDs.)
+PTT_TYPE=SerialPin
+PTT_PORT=/dev/serial/by-id/usb-AIOC_All-In-One-Cable_a3c0b2d0-if04
+PTT_PIN=DTR!RTS
```