skuep/AIOC #88: USB EMI Suggestion

stretchkerr, 2024-10-07:
Regarding the EMI being induced on the USB cable, have you considered swapping the positions of the 2.5mm and 3.5mm jacks, so the USB port as at the bottom of the board, and the USB cable faces down, away from the antenna?

Tymek, 2024-10-26:
Hello, SP5CMD here. I organized a group buy in 🇵🇱, and I got some feedback. USB interference was common. It can be mitigated with higher quality cables. This is because if you're not holding a _handheld_ radio, with an unbalanced antenna, your cable will become a counterpoise. In this situation chokes on cables help.

My sources:
- [gist.github.com/olgierd/de8cdeff16078672d26c6efca8c29971](https://gist.github.com/olgierd/de8cdeff16078672d26c6efca8c29971)
- _Bydgoskie Próby Radiowe_ / SQ2CPA
[image]

---

What I will try, is attaching AIOC close to the source. Maybe for v2 we should have a male USB-C option?

[image]

skuep, 2024-10-27:
Sorry for the late reply... 

In the past there have been some suggestions to "turn around" the AIOC so that the USB cable points downwards. I checked and the effort for re-routing the PCB would have been big unfortunately. I am also not convinced that this helps the EMI issues.

I think the issue is exactly what @Tymek describes but I have not yet found a neat solution for the problem (mainly due to low feedback from others and I don't have any EMI issues myself unfortunately). I was wondering whether a ferrite core on the USB lead is enough to stop the counter-poise current running along the USB cable shield but others didn't have great success. 

Other than that, I was thinking to add SMD ferrites to (some of) the signal lines going to the radio but I am so far not sure, which ones and whether it needs to be a common-mode ferrite (like in @Tymek picture) or if several separate ferrites would work too.

So all in all, it's currently inconclusive from my side on what to do about it.

db4rne, 2025-01-28:
Hi, I'm not sure if this was discussed already, but maybe we shouldn't tie USB shield directly to GND? I tried googling a bit and all the possible versions of it are recommended by someone, good summary is provided here:  https://forum.allaboutcircuits.com/threads/usb-device-cable-shield-connection-grounding-it-or-not.58811/

in short: nobody really seems to have _the_ solution, though most Appnotes seem to agree that just connecting GND and shield is not the way to go. Maybe this is something for the next version to consider?

FFAMax, 2026-10-24:
Hey folks.

Long time ago I found issues with RP2040 connected via 'some' cable via 'come hubs'.
Since that time I had solid short cable for programming purpose connected directly to iMac.
Today I assembled AIOC and found first stramge thing - if plug AIOC, turn radio (Quan Sheng) then screen stay dark, flash light is lighting.
If turn radion on, plug AIOC - it will turn DISABLED sign and in general it will work.
Later, due solif and havy cable, I was afreid to damage AIOC and connectors, so I took longer and thiner cable. Start getting surpizes. One of them 
```

[2003452.313024] usb 1-1.2: [18] FU [AIOC Audio In Capture Volume] ch = 1, val = -24576/0/1
[2003452.315131] hid-generic 0003:1209:7388.002A: hidraw5: USB HID v1.00 Device [AIOC All-In-One-Cable] on usb-0000:00:14.0-1.2/input3
[2003452.315802] cdc_acm 1-1.2:1.4: ttyACM0: USB ACM device
[2003489.900662] usb 1-1-port2: disabled by hub (EMI?), re-enabling...
[2003489.900831] usb 1-1.2: USB disconnect, device number 105
[2003490.073706] usb 1-1.2: new full-speed USB device number 106 using xhci_hcd
[2003490.152398] usb 1-1.2: New USB device found, idVendor=1209, idProduct=7388, bcdDevice= 1.00
[2003490.152403] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[2003490.152405] usb 1-1.2: Product: All-In-One-Cable
[2003490.152406] usb 1-1.2: Manufacturer: AIOC
[2003490.152407] usb 1-1.2: SerialNumber: 850b87f4
[2003490.176145] usb 1-1.2: Warning! Unlikely big volume range (=24576), cval->res is probably wrong.
[2003490.176150] usb 1-1.2: [2] FU [AIOC Audio Out Volume Playback Volume] ch = 1, val = -24576/0/1
[2003490.178699] usb 1-1.2: Warning! Unlikely big volume range (=24576), cval->res is probably wrong.
[2003490.178702] usb 1-1.2: [2] FU [AIOC Audio Out Volume Playback Volume] ch = 1, val = -24576/0/1
[2003490.182434] usb 1-1.2: Warning! Unlikely big volume range (=24576), cval->res is probably wrong.
[2003490.182438] usb 1-1.2: [18] FU [AIOC Audio In Capture Volume] ch = 1, val = -24576/0/1
[2003490.184220] usb 1-1.2: Warning! Unlikely big volume range (=24576), cval->res is probably wrong.
[2003490.184224] usb 1-1.2: [18] FU [AIOC Audio In Capture Volume] ch = 1, val = -24576/0/1
[2003490.186842] hid-generic 0003:1209:7388.002B: hidraw5: USB HID v1.00 Device [AIOC All-In-One-Cable] on usb-0000:00:14.0-1.2/input3
[2003490.187476] cdc_acm 1-1.2:1.4: ttyACM0: USB ACM device
[2003492.972714] usb 1-1-port2: disabled by hub (EMI?), re-enabling...
[2003492.972870] usb 1-1.2: USB disconnect, device number 106
[2003494.786486] usb 1-1-port2: attempt power cycle
[2003496.794575] usb 1-1-port2: unable to enumerate USB device
```

All was fine untill I start trying turn radio to TX. If fall back to solid short cable - all works.
Need troubleshoot firther.

FFAMax, 2026-10-24:
Well, somehow one transistor was fried on my version of PCB (https://github.com/skuep/AIOC/issues/148)